tenbaht/sduino #6: handling of SerialEvent() is not implemented

tenbaht, 2017-02-28:
problem is the lack of a __weak attribute in the compiler. No idea how to allow for a user-defined function that is only called if it exists without generating a linker error.

gicking, 2017-04-29:
hi Michael,
how about using SerialEvent() to change a function pointer which is called in main(). By default this is initialized with a pointer to a dummy function. For example see TIM4UPDAttachInterrupt(fct) in [here](https://github.com/gicking/STM8_templates/blob/master/Library/Base/timer4.c), which allows calling an optional user function call in the 1ms TIM4-ISR.
Hope this helps!?
Regards, Georg

gicking, 2017-04-29:
In the same context: how can I "overload" existing functions with own code, e.g. like SoftwareSerial or the above mentioned optional user code call in existing ISR's?
Any help is highly appreciated!
Regards, Georg

tenbaht, 2017-09-19:
Hi gicking,

it is a while since your suggestion, but now I am back on this project.

This function pointer mechanism would require using a function like "attachSerialEvent()" to initialize this pointer to the non-default value. That is possible, but quite far away from the Arduino-Syntax.

It would be great if I could do something like

```
void (SerialEvent*)(void) = NULL;

main()
{
  setup();
  while (1) {
    loop();
    if (SerialEvent) SerialEvent();
  }
}

```
But defining a function called SerialEvent() would trigger a linker error about duplicate symbols. I don't see any way without a `__weak__` compiler attribute.

gicking, 2017-09-21:
hi tenbaht,

welcome back  ;-)

ok, understood. Without overloading it would be possible only by non-Arduino syntax. However, below is a response by Maarten Brock on this topic -> if you don't compile the source but link to a library containing a dummy SerialEvent() it should work (same as for putchar()). I believe it's not worth the hassle, but I wanted to point it out to you.

Georg
---------------------
From Sdcc-user Newsletter Digest, Vol 128, Issue 2 

'Overloading' can only be done for functions that reside in a library. The linker will first link all sources (actually .rel files a.k.a. objects) it was given and after that it will search missing symbols in the libraries.
So if you provide getchar() in your sources there is no need to search for it in the libraries and yours will be used.

This also means that all code in your object files is linked in whether it's used or not. And objects in the libraries that are not referenced are not linked in. Also note that all functions in an object are linked in,
not just the missing one.

tenbaht, 2017-09-22:
Thank you very much for this hint!

I tried that before it didn't work because of a flaw in the way the makefile links the core library and main.c. Now it is fixed and everything works as expected for me. There is a new test example in test/serialEvent to use the new function.

closed 2017-09-22 by tenbaht

gicking, 2017-09-22:
hi tenbaht,

sound good :-)   Unfortunately I am currently bound in another project, so I won't be able to check it short notice, sorry! 

Nevertheless, keep up the good work!

Georg

PS: I noticed that you use my STM8S SPL patch...? It may be of interest to you that I recently uploaded a major [update of this patch](https://github.com/gicking/STM8-SPL_SDCC_patch)

commit abde8b01 (2017-09-22): fix #6: implement serialEvent() as a 'weak' function