tenbaht/sduino #7: compile errors

gicking, 2017-03-19:
hi,
wanted to try sduino, but I am running into problems even with Blinky :-(   Specifically 

- double missing:
  - compiler warns repeatedly "type 'double' not supported assuming 'float'"
  - linker reports an error "?ASlink-Warning-Definition of public symbol '__divulong' found more than once"
  - however, an IHX is created, but since I don't have that board I couldn't check the functionality
  - is there a way to have double support in SDCC? If not, is it possible to somehow replace double globally by float...?

- STM8 Discovery
  - since I do have a STM8SDiscovery board I tried compiling Blinky for that board
  - compiler aborts with many errors like  
`stm8sdisco/pins_arduino.h:370: error 20: Undefined identifier 'TIMER31'`

Here's my setup:
- Ubuntu 16.04
- sdcc fresh install as described in the Wiki

For your support thanks a lot in advance!

closed 2017-04-25 by tenbaht

tenbaht, 2017-04-25:
Thank you for filing this problem! It seems to be the same problem as with issue #8  . Could you please confirm if it works now for you? Please reopen if not!

gicking, 2017-04-25:
hello,
now example "Blinky" compiles and runs for STM8S Discovery :-)
However I still get warnings for double

tenbaht, 2017-04-26:
Yes, thats a known issue. SDCC doesn't support double. And AFAIK this is not planned for the future.

I was considering adding a `#define double float`. but that might be confusing in case somebody really relies on double precision. You could fix this yourself by adding this line to your Makefile:

```
CPPFLAGS=-Ddouble=float
```

It would be possible to edit the arduino libs as well, but I try to keep the Arduino files as close to the original as possible. Maybe it would be worth giving up on that goal?

commit eafec874 (2017-04-25): fix #7: compile error with timer definitions