tenbaht/sduino #8: OSX 10.10.2 cannot compile Blink for stm8sdisco

andymsheen, 2017-03-27:
I can't seen to compile Blink for stm8sdiscovery board, although seems to compile for stm8sblue. 

Getting these errors:

```
/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c -o build-stm8sdisco/core/wiring_analog.c.rel
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/Print.h:64: warning 93: type 'double' not supported assuming 'float'
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/Print.h:74: warning 93: type 'double' not supported assuming 'float'
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:125: error 20: Undefined identifier 'TIMER11'
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:125: error 62: 'case' expression not constant. Statement ignored
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:151: error 20: Undefined identifier 'TIMER12'
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:151: error 62: 'case' expression not constant. Statement ignored
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:280: error 20: Undefined identifier 'TIMER31'
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:280: error 62: 'case' expression not constant. Statement ignored
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:289: error 20: Undefined identifier 'TIMER32'
/Users/Andy/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino/wiring_analog.c:289: error 62: 'case' expression not constant. Statement ignored
make: *** [build-stm8sdisco/core/wiring_analog.c.rel] Error 1
```

tenbaht, 2017-04-25:
Thank you for filing this problem! It seems to be the same problem as with issue #7 . I just fixed it. As I don't have an OSX system available, could you please confirm if it works now for you? Please reopen if not!

closed 2017-04-25 by tenbaht

andymsheen, 2017-05-08:
worked!