tenbaht/sduino #10: Build for Windows

Jeka4tm, 2017-05-28:
Hey. I could not install until the end. The board is installed, but gives an error

Warning: Board sduino:stm8:stm8sblue doesn't define a 'build.board' preference. Auto-set to: STM8_STM8SBLUE
Warning: Board sduino:stm8:stm8sdisco doesn't define a 'build.board' preference. Auto-set to: STM8_STM8SDISCO
recipe.preproc.macros не найден шаблон

Could collect the necessary files from the folder in order to simply copy the whole to the folder Arduino and it all worked

tenbaht, 2017-09-20:
Sorry, but I am not sure, if I understand you correctly.

These warnings are not generated by the included make files. Did you try to install the sduino-files within your regular Arduino-installation and try to run it from the IDE? But what do you mean by "collecting the necessary files"? Did you finally managed to run it within the IDE? 

IMHO that can't work as it needs a totally different compiler as the regular Arduino platform. It only works from the command line. But if you did find a way to make it possible than I would be **very** interested to hear about the details and include it in the repository!

closed 2017-09-20 by tenbaht

Jeka4tm, 2017-09-20:
Yes, I wanted from among the Arduino ide as STM32