tenbaht/sduino #11: compile example code error in sdcc3.6.0

segatecm, 2017-05-29:
Hi, i download the sduino code and compile the Blink code, it report some errors:

/home/cm/Documents/stm8/sduino-master/sduino/hardware/sduino/stm8/cores/sduino/Print-float.c:59: error 101: too many parameters 
/home/cm/Documents/stm8/sduino-master/sduino/hardware/sduino/stm8/cores/sduino/Print-float.c:60: error 101: too many parameters 
/home/cm/Documents/stm8/sduino-master/sduino/hardware/sduino/stm8/cores/sduino/Print-float.c:101: warning 93: type 'double' not supported assuming 'float'
/home/cm/Documents/stm8/sduino-master/sduino/Arduino.mk:1309: recipe for target 'build-stm8sblue/core/Print-float.c.rel' failed
make: *** [build-stm8sblue/core/Print-float.c.rel] Error 1

my system is:
Linux version 4.4.0-53-generic (buildd@lcy01-20) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #74-Ubuntu SMP Fri Dec 2 15:58:04 UTC 2016

sdcc 3.6.0

evildave666, 2017-06-25:
Lines 59 and 60 in Print-float.c can be commented out, and combined with CPPFLAGS=-Ddouble=float this will compile with sdcc.

segatecm, 2017-06-26:
Yes, it works. thanks.
So the other question, when the new version will be release?

closed 2017-07-03 by segatecm

tenbaht, 2017-09-19:
The latest commit fixes the `'double' not supported` compiler warnings.

And in general I highly recommend using the sdcc snapshot builds instead of the official releases. The compiler code quality for the STM8 has improved a lot since the last release.