tenbaht/sduino #12: Compile error: Various examples

KenjutsuGH, 2017-06-14:
```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = MAC 
- [USER]               ARDUINO_DIR = /Users/k/Documents/sduino/sduino 
- [USER]               ARDMK_DIR = /Users/k/Downloads/sduino-master-2/sduino 
- [AUTODETECTED]       ARDUINO_VERSION = 160 
- [USER]               ARCHITECTURE = stm8 
- [DEFAULT]            OBJSUFFIX = rel 
- [USER]               ARDMK_VENDOR = sduino 
- [USER]               ARDUINO_SKETCHBOOK = /tmp	 
- [AUTODETECTED]       AVR_TOOLS_DIR = / (found in $PATH)
- [COMPUTED]           ARDUINO_LIB_PATH = /Users/k/Documents/sduino/sduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /Users/k/Documents/sduino/sduino/hardware/sduino/stm8/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /Users/k/Documents/sduino/sduino/hardware/sduino/stm8/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /Users/k/Documents/sduino/sduino/hardware/sduino/stm8/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /tmp	/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = stm8sblue 
- [COMPUTED]           CORE = sduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-stm8sblue (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /Users/k/Documents/sduino/sduino/hardware/sduino/stm8/cores/sduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [USER]               MCU_FLAG_NAME = mstm8 -D 
expr: syntax error
expr: syntax error
- [DEFAULT]            CFLAGS_STD =  
expr: syntax error
- [DEFAULT]            CXXFLAGS_STD =  
expr: syntax error
expr: syntax error
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
-
-                      ARDUINO_LIBS =
- [USER]                 /tmp
- [PLATFORM]             LiquidCrystal
- [COMPUTED]           BOOTLOADER_PARENT = /Users/k/Documents/sduino/sduino/hardware/sduino/stm8/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.6.6 #9921 (Mac OS X i386) published under GNU General Public License (GPL) (avr-gcc)

```
```
/Users/k/Downloads/sduino-master-2/sduino/hardware/sduino/stm8/cores/sduino/xmacro.h:217: warning 112: function 'Print_printFloatln' implicit declaration
/Users/k/Downloads/sduino-master-2/sduino/hardware/sduino/stm8/cores/sduino/xmacro.h:217: error 101: too many parameters 
/Users/k/Downloads/sduino-master-2/sduino/hardware/sduino/stm8/cores/sduino/xmacro.h:217: warning 85: in function instance_printFloatln unreferenced function argument : 'arg1'
/Users/k/Downloads/sduino-master-2/sduino/hardware/sduino/stm8/cores/sduino/xmacro.h:217: warning 85: in function instance_printFloatln unreferenced function argument : 'arg2'
```
**sduino/hardware/sduino/stm8/libraries/PCD8544/examples/HelloWorld**
```
HelloWorld.c:53: warning 112: function 'Print_print_f' implicit declaration
HelloWorld.c:53: error 101: too many parameters 
HelloWorld.c:53: warning 85: in function lcd_print_f unreferenced function argument : 'arg1'
HelloWorld.c:53: warning 112: function 'Print_println_f' implicit declaration
HelloWorld.c:53: error 101: too many parameters 
HelloWorld.c:53: warning 85: in function lcd_println_f unreferenced function argument : 'arg1'
```

**sduino/hardware/sduino/stm8/libraries/LiquidCrystal/examples/Blink**
```
Blink.c:48: warning 112: function 'Print_print_f' implicit declaration
Blink.c:48: error 101: too many parameters 
Blink.c:48: warning 85: in function lcd_print_f unreferenced function argument : 'arg1'
Blink.c:48: warning 112: function 'Print_println_f' implicit declaration
Blink.c:48: error 101: too many parameters 
Blink.c:48: warning 85: in function lcd_println_f unreferenced function argument : 'arg1'
```

segatecm, 2017-06-19:
Hi KenjutsuGH, you have same Error like me. So do you resolve the question?

KenjutsuGH, 2017-06-20:
Hi. Unfortunately, I have not been able to solve the problem.

_Sent from my HUAWEI hi6210sft using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

KenjutsuGH, 2017-06-23:
I am getting the same errors under Ubuntu 16.04 LTS

tenbaht, 2017-06-23:
Thank you for the feedback. Unfortunatly, I am quite busy at work in the morment, but I will have a look at this issue next week. Your input is not lost...

KenjutsuGH, 2017-06-23:
@tenbaht Thank you :-)

evildave666, 2017-06-24:
I actually did work this out a couple weeks ago by CPPFLAGS=-Ddouble=float in the makefile and commenting lines 59 & 60 of sduino/sduino/hardware/sduino/stm8/cores/sduino/Print-float.c

KenjutsuGH, 2017-06-24:
Unfortunately, commenting out the two lines did not help :-(

_Sent from my HUAWEI hi6210sft using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Jugulaire, 2017-09-15:
Same right here. The problem come from SDCC. 
Can anybody tell me which version to use for getting things working ?

Jugulaire, 2017-09-15:
Okay, so after compiling myself SDCC, the problem still here : 
```
make                                                                                                 
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/jugu/Arduino/sduino/sduino 
- [USER]               ARDMK_DIR = /home/jugu/Arduino/sduino/sduino 
- [AUTODETECTED]       ARDUINO_VERSION = 160 
- [USER]               ARCHITECTURE = stm8 
- [DEFAULT]            OBJSUFFIX = rel 
- [USER]               ARDMK_VENDOR = sduino 
- [USER]               ARDUINO_SKETCHBOOK = /tmp	 
- [AUTODETECTED]       AVR_TOOLS_DIR = /usr (found in $PATH)
- [COMPUTED]           ARDUINO_LIB_PATH = /home/jugu/Arduino/sduino/sduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/boards.txt (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /tmp	/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = stm8sblue 
- [COMPUTED]           CORE = sduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-stm8sblue (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/cores/sduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [USER]               MCU_FLAG_NAME = mstm8 -D 
expr: syntax error
expr: syntax error
- [DEFAULT]            CFLAGS_STD =  
expr: syntax error
- [DEFAULT]            CXXFLAGS_STD =  
expr: syntax error
expr: syntax error
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
-
-                      ARDUINO_LIBS =
- [USER]                 /tmp
- [PLATFORM]             PCD8544
- [COMPUTED]           BOOTLOADER_PARENT = /home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.6.9 #9995 (Linux) published under GNU General Public License (GPL) (avr-gcc)
-------------------------
../../sduino.mk:111: warning: overriding recipe for target 'upload'
/home/jugu/Arduino/sduino/sduino/Arduino.mk:1512: warning: ignoring old recipe for target 'upload'
mkdir -p build-stm8sblue
/opt/sdcc/bin/sdcc "-Wp-MMD build-stm8sblue/lcd.c.d" -c -I. -I/home/jugu/Arduino/sduino/sduino/../STM8S_StdPeriph_Driver/inc -I/opt/sdcc/share/sdcc/include/ -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=160 -DARDUINO_ARCH_STM8 -D__PROG_TYPES_COMPAT__ -I/home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/cores/sduino -I/home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/variants/standard    -I/home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/libraries/PCD8544   -I/tmp   lcd.c -o build-stm8sblue/lcd.c.rel
/home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/cores/sduino/Print.h:64: warning 93: type 'double' not supported assuming 'float'
/home/jugu/Arduino/sduino/sduino/hardware/sduino/stm8/cores/sduino/Print.h:74: warning 93: type 'double' not supported assuming 'float'
lcd.c:4: warning 93: type 'double' not supported assuming 'float'
lcd.c:4: warning 112: function 'Print_print_f' implicit declaration
lcd.c:4: error 101: too many parameters 
lcd.c:4: warning 85: in function lcd_print_f unreferenced function argument : 'arg1'
lcd.c:4: warning 93: type 'double' not supported assuming 'float'
lcd.c:4: warning 93: type 'double' not supported assuming 'float'
lcd.c:4: warning 112: function 'Print_println_f' implicit declaration
lcd.c:4: error 101: too many parameters 
lcd.c:4: warning 85: in function lcd_println_f unreferenced function argument : 'arg1'
lcd.c:4: warning 93: type 'double' not supported assuming 'float'
lcd.c:4: warning 112: function 'Print_printFloatln' implicit declaration
lcd.c:4: error 101: too many parameters 
lcd.c:4: warning 85: in function lcd_printFloatln unreferenced function argument : 'arg1'
lcd.c:4: warning 85: in function lcd_printFloatln unreferenced function argument : 'arg2'
/home/jugu/Arduino/sduino/sduino/Arduino.mk:1263: recipe for target 'build-stm8sblue/lcd.c.rel' failed
make: *** [build-stm8sblue/lcd.c.rel] Error 1

```

tenbaht, 2017-09-19:
The latest commit 7cc64c0acedb94e91f3922d25b54a4799b812af4 should fix this issue. It was not a compiler problem, just simple inconsistent function naming within the source files. I fixed the basic print test in test/print and I think that there aren't any more files using the wrong syntax anymore.

@KenjutsuGH, @segatecm and @Jugulaire: could you please confirm that the problem is fixed?

Sorry, that it took me so long to respond. Hopefully I will have more time for this project again in the next few months.

closed 2017-09-19 by tenbaht

Jugulaire, 2017-09-20:
Hey, 

Thank you for this fix ! Works like charm on the one dollar dev board. 
Tested with Nokia screen for the moment but the liquid crystal should work too !

KenjutsuGH, 2017-09-22:
Thank you @tenbaht!

It now works like a charm :-) Just awesome to see the little **STM8** board driving a **Nokia LCD**