tenbaht/sduino #14: "Notes on SDCC" on website

spth, 2017-07-16:
The project website has a section "Notes on SDCC". It mentions a few issues:

1 Directly connected sequences of 'addw x,#' and 'subw x,#' should be combined into one operation.
2 Multiplication by two is done by 'mul' instead of a bitshift. Important for array access.
3 Interrupt routine preamble: Why is there a 'clr a/div x,a' sequence?
4 Indirect 16 bit access: 'ldw x,#addr/ldw x,(x)' should be 'ldw x,[addr]'
5 Indirect function call: 'ldw x,#addr/ldw x,(x)/call (x)' should be 'call [addr]'

Do you have some compileable C code that shows 1 and 2? I'm not aware of any such major issues with SDCC.

3: The interrupt routine contains a call through a function pointer, which makes it impossible for SDCC to know which instructions will be in the ISR. The div x, a has an undocumented side-effect of clearing an undocumented bit in the flag register, which, depending on the instructions in the ISR is a necessary hardware-bug workround (see "Unexpected DIV/DIVW instruction result in ISR" in the STM8 errata, which suggest a different workaround).
4: According to the STM8 manual, 'ldw x,[addr]' while only 1 instruction actually uses as many cycles as the 2 instructions generated by SDCC and is 1 byte longer.
5: According to the STM8 manual, 'call [addr]' while only 1 instruction actually uses as many cycles and bytes as the 2 instructions generated by SDCC.

Philipp

P.S.: I've implemented a few more optimizations in SDCC (and the stm8backend) recently. If you have a few examples where current SDCC does generate inefficient code, I might be able to have a look at it.

tenbaht, 2017-09-20:
Hi Philipp,

## issue 1
Unfortunately, I didn't keep the files when I found that problem. I can't find any examples anymore, so I assume it was fixed in one of the later compiler revisions.

## issue 2
minimal example code:
```
int array[4];
void array_access(unsigned char index)
{
    array[index] = 0;
}
```
compiled using the latest snapshot build 9998:
```
$ /opt/sdcc/bin/sdcc -c -mstm8 issue14-2.c
$ /opt/sdcc/bin/sdcc -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.6.9 #9998 (Linux)
published under GNU General Public License (GPL)
```
results in this code:
```
      000000                         47 _array_access:
      000000 52 02            [ 2]   48 	sub	sp, #2
                                     49 ;	issue14.c: 6: array[index] = 0;
      000002 AEr00r00         [ 2]   50 	ldw	x, #_array+0
      000005 1F 01            [ 2]   51 	ldw	(0x01, sp), x
      000007 7B 05            [ 1]   52 	ld	a, (0x05, sp)
      000009 97               [ 1]   53 	ld	xl, a
      00000A A6 02            [ 1]   54 	ld	a, #0x02
      00000C 42               [ 4]   55 	mul	x, a
      00000D 72 FB 01         [ 2]   56 	addw	x, (0x01, sp)
      000010 6F 01            [ 1]   57 	clr	(0x1, x)
      000012 7F               [ 1]   58 	clr	(x)
      000013 5B 02            [ 2]   59 	addw	sp, #2
      000015 81               [ 4]   60 	ret
```
When I look at it I see that the `mul` command already includes the transfer of the result into x which needs to happen anyway. It would be great if the temporary variable (0x01,sp) could be avoided in the first place.

## issues 3-5
Thank you for your explanation!

I noticed the recent changes in the backend, the optimizer and the simulator. I would like to compare the development of the code size compared to an older version. Thank you for all the work there!

  Michael

spth, 2017-09-20:
Thanks. I can see how using sllw could save 1 byte and 2 cycles vs the current solution using mul. I'll probably look into implementing it next week.

Philipp

spth, 2017-09-21:
As of revision #10000, a sllw is used instead of the mul. The mcs51 backend can optimize out the temporary that holds the array address; I'll look into porting the solution to stm8 (and maybe z80).

Philipp

tenbaht, 2017-10-24:
sllw works with rev. 10088 and saves one byte.

I am looking forward to see the temporary optimized out, that could save 7 bytes on every array access. I will follow the discussion on SF, no need to duplicate it here.

closed 2017-10-24 by tenbaht