tenbaht/sduino #16: Compile error ubuntu 14, sdcc 3.4.0

colesnicov, 2017-09-16:
Hi everyone.

I can not compile an example of blinking. I'm getting an error:

```
/home/denis/git/sduino/sduino/hardware/sduino/stm8/cores/sduino/Print.h:64: warning 93: type 'double' not supported assuming 'float'
/home/denis/git/sduino/sduino/hardware/sduino/stm8/cores/sduino/Print.h:74: warning 93: type 'double' not supported assuming 'float'
/home/denis/git/sduino/sduino/hardware/sduino/stm8/cores/sduino/Arduino.h:294: syntax error: token -> 'unsigned' ; column 15
make: *** [build-stm8sblue/Blink.c.rel] Error 1
```

I have Ubuntu 14 and sdcc verion 3.4.0

Thanks.

Jugulaire, 2017-09-16:
STM8 are not supported on version older than 3.5.0

colesnicov, 2017-09-17:
Yes, you right. SDCC 3.6.9 works accurately...

closed 2017-09-17 by colesnicov