tenbaht/sduino #19: Request: Support for / Opinions on sduino UNO board

roybaer, 2017-10-07:
I spent the last couple of weeks designing this [sduino UNO board](https://github.com/roybaer/sduino_uno) (hope you are okay with me shamelessly borrowing the name).
Now I wanted to ask for opinions on the pin mapping I came up with.

My approach was to try and find as good a function mapping as possible while keeping everything that used to be in a particular AVR port register in one STM8 port register, since this allows for more compatible parallel IO.

It would especially be nice to find a harmonized pin mapping for this STM8S105K6 board and the STM8S105K4 board mentioned in Issue #17 before I get the PCBs manufactured and to hear whether you think that anything about my current pin mapping looks like a silly idea.

tenbaht, 2017-10-07:
Whow, that looks amazing! I am looking forward to see the first prototypes!

- I really like the pin mapping. I don't see a way to get closer to the original mapping than this. Used like this, the functional mapping makes sense.
- Implementing the serial interface clearly calls for bootloader support. Somewhere I have read about such a project, but I never looked into it. It would be extremely useful to integrate that here.

Additionally, I wrote some comments regarding the hardware design to your repository.

tenbaht, 2017-11-20:
integrated by commit 6589a0aa9ec6046e868ed0ee0777637efefa03ba

closed 2017-11-20 by tenbaht