tenbaht/sduino #22: Bootloader support via stm8gal

roybaer, 2017-10-22:
Bootloader support would be nice for those devices with embedded ROM bootloader. [stm8gal](https://github.com/gicking/stm8gal.git) looks promising. It supports resetting the board via DTR line, as well.

tenbaht, 2017-10-23:
I am already working on it. Hopefully it won't take long. The binaries are already part of my next commit I am preparing right now.

roybaer, 2017-10-29:
I've been able to program an (almost) fully assembled sduino UNO board with this line:
`./stm8gal -p /dev/ttyUSB0 -R 1 -u 2 -Q -w ~/sduino/sduino/examples/01.Basics/Blink/build-s8uno/Blink.ihx`
Once I figured out the `-u 2` part everything went smoothly.

Annoyingly, it's the LEDs that have not arrived, yet, so I had to test the Blink demo with an external one.

tenbaht, 2017-11-20:
6589a0aa9ec6046e868ed0ee0777637efefa03ba integrates bootloader support for the IDE. It works for a board manager install, a manual install still needs a different path definition in platform.txt.

So this is kind of half-closed now.

dubelectro, 2017-11-22:
I built stm8gal for macOS.
I uploaded the macOS version below.
https://goo.gl/5GaXWE

I modified board.txt on my own and tested it with sduino.
It seems to be working correctly.
I tried it with stm8sblack (stm8s105k4) and stm8s discovery.

tenbaht, 2017-11-22:
On Wed, 22 Nov 2017 12:20:24 +0000 (UTC)
dubelectro <notifications@github.com> wrote:

> I built stm8gal for macOS.
> I uploaded the macOS version below.
> https://goo.gl/5GaXWE

Great! Thank you very much! It will be included with v0.4.0

> I modified board.txt on my own and tested it with sduino.

On Monday I pushed a commit to add stm8gal support for the 105. Did you
use that version as a base or did you start with the 0.3.3?

The current master now included stm8gal and the new platform.txt. It
works for a board manager install, but I didn't pack a release. For a
manual install it will show the problem with the tools path for upload.

> It seems to be working correctly.
> I tried it with stm8sblack (stm8s105k4) and stm8s discovery.

Did you add stm8sblack as a new board? And which pinout did you use?
Maybe it is useful to add it to the repository?

Thank you for your help to get the Mac version working!

  Michael

dubelectro, 2017-11-24:
> On Monday I pushed a commit to add stm8gal support for the 105. Did you
> use that version as a base or did you start with the 0.3.3?

I based on a committed version. For stm8sblack, I did as follows.
```
menu.upload_method=Upload method
##############################################################
stm8sblack.name=STM8S105K4T6 Breakout Board

stm8sblack.menu.upload_method.stm8flashMethod=stm8flash(STlinkV2)
stm8sblack.menu.upload_method.stm8flashMethod.upload.tool=stm8flash
stm8sblack.menu.upload_method.stm8flashMethod.upload.protocol=stlinkv2
stm8sblack.menu.upload_method.stm8flashMethod.upload.mcu=stm8s105?4

stm8sblack.menu.upload_method.stm8galMethod=stm8gal(Serial)
stm8sblack.menu.upload_method.stm8galMethod.upload.tool=stm8gal
stm8sblack.menu.upload_method.stm8galMethod.upload.protocol=serial
stm8sblack.menu.upload_method.stm8galMethod.upload.speed=230400
stm8sblack.upload.maximum_size=16384
stm8sblack.upload.maximum_data_size=2048

stm8sblack.build.mcu=STM8S105
stm8sblack.build.f_cpu=16000000L
stm8sblack.build.board=STM8S_BLACK
stm8sblack.build.core=sduino
stm8sblack.build.variant=stm8sblack

# meaningless variables just to keep the makefile happy

stm8sblack.upload.speed=1
```

> The current master now included stm8gal and the new platform.txt. It
> works for a board manager install, but I didn't pack a release. For a
> manual install it will show the problem with the tools path for upload.
> 

It seems to be better to follow the following method for manual installation.
https://github.com/stm32duino/wiki/wiki/Using-git-repository
It is necessary to install sduino once with the board manager, but the path problem will probably disappear.

> Did you add stm8sblack as a new board? And which pinout did you use?
> Maybe it is useful to add it to the repository?

I modified the definition of s8uno which is the same peripheral first and tried it. Next, I created a variant for stm8sblack. I put the files in the following places.
https://github.com/dubelectro/sduino/tree/master/sduino/hardware/sduino/stm8/variants/stm8sblack
This is incomplete. Because I have no programming skills. I have some unclear points. Please correct if there is a problem.

And there is even more problem. To use the altanative function of stm8, you need to change option byte. Defining it with variant seems to be insufficient, and some functions may not work. For example, I2C of stm8black (probably also s8uno) is altanative function.  Is there a good way?
 I changed a option byte on Windows using svp. I think it can be changed in sketch, but I do not know how to do it.

tenbaht, 2017-12-06:
My 105 boards finally arrived and I started to integrate your suggestions. I hope I will find enough time to finish that soon.

I put the whole idea of using the alternate functions on hold. The idea of Arduino is simpicity. I am afraid that alternate functions would add to much complexity to the system. A different pinout depending on the state of some invisible option bytes might be more confusing than useful in most cases.

If anybody really needs to use alternate functions, it is still possible to do so using additional code in user space.

tenbaht, 2017-12-06:
I tried to integrate your definition for stm8sblack. I posted some notes about this into the discussion of issue #17

gicking, 2018-09-30:
hi Michael,
just stumbled accross sduino (again) and noticed that you now include [stm8gal](https://github.com/gicking/stm8gal.git) as upload tool. Seems like my effort was not pointless  :-)

Please note that [stm8gal](https://github.com/gicking/stm8gal.git) is currently at v1.1.7, while the bundled (linux) version is v1.1.3. There were several bugfixes. If it helps I am happy to provide binaries for Linux 32 & 64 bit, Windows and Raspbian 1 & 3. Would it help to add the binaries to my Github repo? 

Regards,
Georg (gicking) 

PS: same applies for sdcc (3.6.9 vs. 3.8.0), while stm8flash seems to be up to date (v1.1)

gicking, 2018-09-30:
hi again,
seems like you are using the [sdcc patch](https://github.com/gicking/STM8-SPL_SDCC_patch) for the SPL by STM...? I didn't check the used patch version, but please note that there have been some bugfixes.

Regards, Georg

tenbaht, 2018-10-08:
thank you for mentioning that. Today is the first day I managed to get back to sduino for a long time. But now our little guy is starting to go to day care and that should help freeing up some time for this project again.
Updating the tools to more current versions will be one of the first things to do. I would highly appreciate binaries for stm8gal and the other tools.

gicking, 2018-10-09:
I know exactly what you mean - been there, done that...  ;-)  But believe me, time flies and your kids will request less and less of your time (sometimes alas...)

Anyway, I will upload Linux64, Linux32 and Windows binaries of stm8gal to the [repo](https://github.com/gicking/stm8gal). As for MacOSX I cannot promise (my brother has a Mac, but I doubt he has installed the required development tools)

tenbaht, 2018-11-21:
commit 52d283ccc closes this.

closed 2018-11-21 by tenbaht