tenbaht/sduino #23: after Board Manager install the Makefile needs manual tweaking

tenbaht, 2017-10-23:
The Board Managers-generated install path for the tools folder is different from the regular pattern. The Makefile needs to auto-detect this instead of relying on the standard paths.

zoomx, 2017-10-24:
So I get this message
bash.exe: warning: could not find /tmp, please create!
for this reason?

tenbaht, 2017-10-24:
No, that message is normal for an IDE build, it doesn't hurt. Not sure how to get rid of it.

Today I wrote instructions on how to get ready for a Makefile. Maybe you could give me some feedback if it is understandable and if it works for you?

https://tenbaht.github.io/sduino/usage/update-to-manual/

zoomx, 2017-10-26:
I tested like a manual install in sketchbook/hardware folder. Win7

Repository downloaded 
Zip unpacked in a folder
Inside this folder there is a sduino-master folder that I renamed to sduino
Copied folders found in packages/sduino/tools in the extracted sduino/tools where there is only "sdcc-critcal.patch" file.
Removed sduino package from Board Manager then I closed the IDE.
Deleted sduino folder from package folder
Moved the extracted sduino folder into /sketchbook/Hardware folder that usually is used for manual core installing
Searched for the sdcc binary that in my portable installation is in
D:\IDE\arduino\portable\sketchbook\Hardware\sduino\tools\sdcc\build.10088\bin

Updated path (on Win7, right click on Computer then Properties, then Advanced Properties, then Environment Variables button, then I modified path only for user). I added the path above.
Opened a cmd.exe and checked with sdcc -v
(Note. I modified path adding the old path were I unpacked sduino. Then updated path but the update didn't work, in cmd the path was the older one. So I disconnected and reentered)
Started IDE I didn't see the STM8 boards.
Maybe it is because boards.txt is here
D:\IDE\arduino\portable\sketchbook\Hardware\sduino\sduino\hardware\sduino\stm8
I made some errors in copying folders :-(

tenbaht, 2017-10-26:
Yes, I see two wrong paths here.
Problem 1: The board manager unpacks the tools package into sduino/tools/sdcc/build.10088, but you want it in tools/sdcc (same for wrapper)

Problem 2: Don't put the whole repository into your Arduino/hardware folder. Only the sduino/hardware folder of the repository belongs there. I recommend setting a link Arduino/hardware/sduino pointing to <repository-dir>/sduino/hardware.

Hope that helps.  I reworked the installation instructions a few times already, but somehow it always sounds confusing.