tenbaht/sduino #24: Board Manager Install, IDE Build not working

speedbug78, 2017-11-01:
First time trying Sduino.

1. Did a board manager install of Sduino per the instructions (running Arduino 1.8.2):

> Start the Arduino-IDE. In File->Preferences, Settings tab, enter

>    https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json
>
>as an Additional Boards Manager URL.
>
>    Open Tools->Board:...->Boards Manager
>    Find Sduino by typing 'sd' into the search line
>    Click on the list entry
>    Click on Install.

2. Attempted to "Verify" stock Blink example (File->Examples->01.Basics->Blink) with both STM8S103F3 and STM8S105C6 boards.  I'm using the Arduino IDE.  Neither worked.

3. Uninstalled 1.8.2 and upgraded to 1.8.5.

4. Attempted again and received the same errors:

`bash.exe: warning: could not find /tmp, please create!
\033[0;36mC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\Blink.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include\033[0m
\033[0;32mMark re12:\033[0m C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\Blink.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
\033[0;31mcpp gefunden\033[0m
bash.exe: warning: could not find /tmp, please create!
\033[0;36mC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\Blink.ino.cpp sketch\Blink.ino.cpp.o re2 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include\033[0m
\033[0;32mMark re2:\033[0m C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\Blink.ino.cpp -o sketch\Blink.ino.cpp.o
\033[0;31mcpp gefunden\033[0m
bash.exe: warning: could not find /tmp, please create!
\033[0;36mC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino\HardwareSerial.c core\HardwareSerial.c.o re1 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include\033[0m
\033[0;32mMark re1:\033[0m C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino\HardwareSerial.c -o core\HardwareSerial.c.o
C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino\HardwareSerial.c:38:2: warning: #warning "using uart1 for HardwareSerial"

bash.exe: warning: could not find /tmp, please create!
\033[0;36mC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino\Print-float.c core\Print-float.c.o re1 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include\033[0m
\033[0;32mMark re1:\033[0m C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino\Print-float.c -o core\Print-float.c.o
bash.exe: warning: could not find /tmp, please create!
\033[0;36mC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino\Print.c core\Print.c.o re1 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include\033[0m
\033[0;32mMark re1:\033[0m C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\U [... cut ...]
?ASlink-Warning-Undefined Global '_UART1_DeInit' referenced by module 'HardwareSerial'

?ASlink-Warning-Undefined Global '_GPIO_DeInit' referenced by module 'wiring'

?ASlink-Warning-Undefined Global '_CLK_GetClockFreq' referenced by module 'HardwareSerial'

?ASlink-Warning-Undefined Global '_ADC1_DeInit' referenced by module 'wiring'

?ASlink-Warning-Undefined Global '_TIM1_TimeBaseInit' referenced by module 'wiring'

?ASlink-Warning-Undefined Global '_TIM2_TimeBaseInit' referenced by module 'wiring'

exit status 1
Error compiling for board STM8S103F3 Breakout Board.
`

tenbaht, 2017-11-01:
The linker does not find the precomiled SPL. The path
`-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8`
should be 
`-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\lib`
I am not sure what happend there. It looks like the linker command got cut off when you copied the log, there should be many more paths following. Could you please try again with the verbose compile setting on and copy the log again? (In Preferences check "Show verbose output during compilation")

And could you please check with the Explorer where the library file STM8S103.lib ended up?

speedbug78, 2017-11-01:
Ok, checked verbose output during compilation and set Compiler warnings to All:

There seems to be an odd characters in that line which prevents windows from copying and pasting it no matter how I highlight and copy.  I managed to get it in by copying and pasting several pieces of the line individually, note that the line brakes are false however.

Just the problem line:
`[0;36m--nostdlib -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8 .3.0/STM8S_StdPeriph_Driver/lib -LC:\Users\WirickD\AppData\Local\Temprduino_build_322840 -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino	ools\sdccuild.10088/lib/stm8 --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 C:\Users\WirickD\AppData\Local\Temprduino_build_322840\sketch\Blink.ino.cpp.o C:\Users\WirickD\AppData\Local\Temprduino_build_322840/..rduino_cache_109928?ASlink-Warning-Couldn't find library 'STM8S103'`

The entire output:
`Arduino: 1.8.5 (Windows 7), Board: "STM8S103F3 Breakout Board"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\WirickD\AppData\Local\Arduino15\packages -hardware C:\Users\WirickD\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\WirickD\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\WirickD\Documents\Arduino\libraries -fqbn=sduino:stm8:stm8sblue -ide-version=10805 -build-path C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840 -warnings=all -build-cache C:\Users\WirickD\AppData\Local\Temp\arduino_cache_109928 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.sdcc.path=C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088 -prefs=runtime.tools.STM8Tools.path=C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\WirickD\AppData\Local\Arduino15\packages -hardware C:\Users\WirickD\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\WirickD\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\WirickD\Documents\Arduino\libraries -fqbn=sduino:stm8:stm8sblue -ide-version=10805 -build-path C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840 -warnings=all -build-cache C:\Users\WirickD\AppData\Local\Temp\arduino_cache_109928 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.sdcc.path=C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088 -prefs=runtime.tools.STM8Tools.path=C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
Using board 'stm8sblue' from platform in folder: C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0
Using core 'sduino' from platform in folder: C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0
Detecting libraries used...
"C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/win/bash" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc.sh" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840\sketch\Blink.ino.cpp" "nul" re12  -c -Ddouble=float -D__PROG_TYPES_COMPAT__  -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8   "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino" "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard" "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc" "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include"
Generating function prototypes...
"C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/win/bash" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc.sh" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840\sketch\Blink.ino.cpp" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840\preproc\ctags_target_for_gcc_minus_e.cpp" re12  -c -Ddouble=float -D__PROG_TYPES_COMPAT__  -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8   "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino" "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard" "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc" "-IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include"
bash.exe: warning: could not find /tmp, please create!
[... cut ...]
?ASlink-Warning-Undefined Global '_UART1_DeInit' referenced by module 'HardwareSerial'

?ASlink-Warning-Undefined Global '_GPIO_DeInit' referenced by module 'wiring'

?ASlink-Warning-Undefined Global '_CLK_GetClockFreq' referenced by module 'HardwareSerial'

?ASlink-Warning-Undefined Global '_ADC1_DeInit' referenced by module 'wiring'

?ASlink-Warning-Undefined Global '_TIM1_TimeBaseInit' referenced by module 'wiring'

?ASlink-Warning-Undefined Global '_TIM2_TimeBaseInit' referenced by module 'wiring'

exit status 1
Error compiling for board STM8S103F3 Breakout Board.`

speedbug78, 2017-11-01:
I found stm8s103.lib where I expected it:

C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\src\stm8s103.lib

speedbug78, 2017-11-01:
Here is a screen shot if it helps:

[image]

tenbaht, 2017-11-02:
STM8S103.lib (uppercase) is supposed to be in C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\lib
The file you found is a leftover from very early stages. This needs a clean up...

I am pretty sure the problem is an accidential interpretation of  'stm8\0.3.0' as 'stm8' '\0' (=ASCII 0) '.3.0', probably a missing quotation in wrapper/sdcc-link.sh

Not sure why I can't reproduce it here... I am working on it...

tenbaht, 2017-11-02:
I tried a physical Windows7 machine with a fresh install of Arduino 1.8.5 and Sduino, but I can't reproduce the issue. Which windows version are you using?

To narrow it down, could you please do two modifications in tools/wrapper/sdcc-link.sh, do a test run after each change and post the output following the line 'Linking everything together...' here?

1. add a line `set -x` just before the line `SDCC="$1"`
2. change `OBJS+=(${FILE})` to `OBJS+=("${FILE}")`

speedbug78, 2017-11-02:
I have Windows 7 Enterprise, SP1, 64bit.  Contents of C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\lib:
[image]

speedbug78, 2017-11-02:
I don't have a tools/wrapper folder.  I have tools/STM8Tools/2017.10.21/wrapper.
On line 24 I now have: 
> set -x SDCC="$1"

`Linking everything together...
"C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/win/bash" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc-link.sh" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc" --nostdlib "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib" "-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_322840" "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8" --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103  "C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840\sketch\Blink.ino.cpp.o" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840/..\arduino_cache_109928\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.a" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840/core/main.c.rel" -lSTM8S103 -lstm8 --out-fmt-elf -o "C:\Users\WirickD\AppData\Local\Temp\arduino_build_322840/Blink.ino.elf"
bash.exe: warning: could not find /tmp, please create!
+ shift

+ echo -e '\033[0;36m\033[0m'
[0;36m[0m
+ declare -a OBJS
+ '[' 0 -gt 0 ']'
+ echo ''
+ ''
C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc-link.sh: line 45: : command not found
exit status 127
Error compiling for board STM8S103F3 Breakout Board.`

speedbug78, 2017-11-02:
When I change line 41 to:
> 	OBJS+=("${FILE}")

`Linking everything together...
"C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/win/bash" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc-link.sh" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc" --nostdlib "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib" "-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301" "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8" --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103  "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.o" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/core\core.a" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/core/main.c.rel" -lSTM8S103 -lstm8 --out-fmt-elf -o "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf"
bash.exe: warning: could not find /tmp, please create!
+ shift
+ echo -e '\033[0;36m\033[0m'

[0;36m[0m
+ declare -a OBJS
+ '[' 0 -gt 0 ']'
+ echo ''
+ ''
C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc-link.sh: line 45: : command not found
exit status 127
Error compiling for board STM8S103F3 Breakout Board.`

speedbug78, 2017-11-02:
Line 45 is:
> "$SDCC" "${OBJS[@]}"
which is echoed on line 44

tenbaht, 2017-11-02:
> I don't have a tools/wrapper folder. I have tools/STM8Tools/2017.10.21/wrapper.

Yes, perfect.

`set -x`and SDCC="$1"` in two separate lines, not combined. That causes all the other problems here.

speedbug78, 2017-11-02:
Sorry about that.  Here it is with set -x on line 24 and SDCC="$1" on line 25

`"C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/win/bash" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc-link.sh" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc" --nostdlib "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib" "-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301" "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8" --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103  "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.o" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/..\arduino_cache_132091\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.a" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/core/main.c.rel" -lSTM8S103 -lstm8 --out-fmt-elf -o "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf"
bash.exe: warning: could not find /tmp, please create!
+ SDCC='C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc'
+ shift--nostdlib

-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib
-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301
-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8
--code-size
8192
--iram-size
1024
-mstm8
-DSTM8S103
C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.o
C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/..\arduino_cache_132091\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.a
+ echo -e '\033[0;36m--nostdlib' '-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib' '-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301' '-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8' --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.o' 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/..\arduino_cache_132091\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.a' 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/core/main.c.rel' -lSTM8S103 -lstm8 --out-fmt-elf -o 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf\033[0m'
[0;36m--nostdlib -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8`

speedbug78, 2017-11-02:
ug, its chopping off the cut and paste again, let me see if I can get a full output.

speedbug78, 2017-11-02:
Here it is minus whatever character is between the stm8 and .3.0:

`Linking everything together...
"C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/win/bash" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21/wrapper/sdcc-link.sh" "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc" --nostdlib "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib" "-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301" "-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8" --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103  "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.o" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/..\arduino_cache_132091\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.a" "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/core/main.c.rel" -lSTM8S103 -lstm8 --out-fmt-elf -o "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf"
bash.exe: warning: could not find /tmp, please create!
+ SDCC='C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc'
+ shift--nostdlib

-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib
-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301
-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8
--code-size
8192
--iram-size
1024
-mstm8
-DSTM8S103
C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.o
C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/..\arduino_cache_132091\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.a
+ echo -e '\033[0;36m--nostdlib' '-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib' '-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301' '-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8' --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.o' 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/..\arduino_cache_132091\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.a' 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/core/main.c.rel' -lSTM8S103 -lstm8 --out-fmt-elf -o 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf\033[0m'
[0;36m--nostdlib -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8 .3.0/STM8S_StdPeriph_Driver/lib -LC:\Users\WirickD\AppData\Local\Temprduino_build_218301 -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino	ools\sdccuild.10088/lib/stm8 --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 C:\Users\WirickD\AppData\Local\Temprduino_build_218301\sketch\Blink.ino.cpp.o C:\Users\WirickD\AppData\Local\Temprduino_build_218301/..rduino_cache_132091+ declare -a OBJS
+ '[' 18 -gt 0 ']'
+ echo --nostdlib
+ FILE=--nostdlib
+ [[ --nostdlib == *.a ]]
+ [[ --nostdlib == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 17 -gt 0 ']'
+ echo '-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib'
+ FILE='-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib'
+ [[ -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib == *.a ]]
+ [[ -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/lib == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 16 -gt 0 ']'
+ echo '-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301'
+ FILE='-LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301'
+ [[ -LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301 == *.a ]]
+ [[ -LC:\Users\WirickD\AppData\Local\Temp\arduino_build_218301 == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 15 -gt 0 ']'
+ echo '-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8'
+ FILE='-LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8'
+ [[ -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8 == *.a ]]
+ [[ -LC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8 == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 14 -gt 0 ']'
+ echo --code-size
+ FILE=--code-size
+ [[ --code-size == *.a ]]
+ [[ --code-size == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 13 -gt 0 ']'
+ echo 8192
+ FILE=8192
+ [[ 8192 == *.a ]]
+ [[ 8192 == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 12 -gt 0 ']'
+ echo --iram-size
+ FILE=--iram-size
+ [[ --iram-size == *.a ]]
+ [[ --iram-size == *.o ]]
+ OBJS+=("${FILE}")
+ shift
[... 72 lines cut ...]
+ shift
+ '[' 2 -gt 0 ']'
+ echo -o
+ FILE=-o
+ [[ -o == *.a ]]
+ [[ -o == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 1 -gt 0 ']'
+ echo 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf'
+ FILE='C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf'
+ [[ C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf == *.a ]]
+ [[ C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf == *.o ]]
+ OBJS+=("${FILE}")
+ shift
+ '[' 0 -gt 0 ']'
[... cut ...] _cache_132091\core\core_sduino_stm8_stm8sblue_c9eb0cdd5bfd5ed20fbb4a97e5354e19.lib' 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/core/main.c.rel' -lSTM8S103 -lstm8 --out-fmt-elf -o 'C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301/Blink.ino.elf'
?ASlink-Warning-Couldn't find library 'STM8S103'

?ASlink-Error-<cannot open> : "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.rel"

exit status 1
Error compiling for board STM8S103F3 Breakout Board.
`

tenbaht, 2017-11-03:
Great, one problem is fixed now!

But something else is really weired here. Somehow your install came out very different from the other windows machines I tested.

Your C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\lib is missing all the symbolic links that are supposed to be there. Which filesystem are you using? FAT32? Not likely with Windows7. But even NTFS supports links now (finally...)
It is supposed to look like this:
```
lrwxrwxrwx 1 mmayer mmayer     12 Okt 27 13:04 STM8AF52Ax.lib -> STM8S208.lib
-rw-rw-r-- 1 mmayer mmayer 288956 Okt 27 13:04 STM8AF622x.lib
lrwxrwxrwx 1 mmayer mmayer     12 Okt 27 13:04 STM8AF626x.lib -> STM8S005.lib
lrwxrwxrwx 1 mmayer mmayer     12 Okt 27 13:04 STM8AF62Ax.lib -> STM8S007.lib
-rw-rw-r-- 1 mmayer mmayer 279194 Okt 27 13:04 STM8S003.lib
-rw-rw-r-- 1 mmayer mmayer 310826 Okt 27 13:04 STM8S005.lib
-rw-rw-r-- 1 mmayer mmayer 319732 Okt 27 13:04 STM8S007.lib
lrwxrwxrwx 1 mmayer mmayer     12 Okt 27 13:04 STM8S103.lib -> STM8S003.lib
lrwxrwxrwx 1 mmayer mmayer     12 Okt 27 13:04 STM8S105.lib -> STM8S005.lib
lrwxrwxrwx 1 mmayer mmayer     12 Okt 27 13:04 STM8S207.lib -> STM8S007.lib
-rw-rw-r-- 1 mmayer mmayer 386838 Okt 27 13:04 STM8S208.lib
-rw-rw-r-- 1 mmayer mmayer 286736 Okt 27 13:04 STM8S903.lib
```
As a quick fix copy the file STM8S003.lib two times to the new names STM8S103.lib and STM8S105.lib, that will fix the linking problem. But why is your installation so different? The links are part of the archive and obviously the archive got unpacked. So where are the links? I have no idea what happened there.

The next strange thing is the missing {build.path} component for the filenames of your sketch. The linker can't find "C:\Users\WirickD\AppData\Local\Temp\arduino_build_218301\sketch\Blink.ino.cpp.rel" because it was build earlier as "sketch\Blink.ino.cpp.rel", without the path before it. But why? This filename is generated by the Arduino IDE and used by arduino-builder with platform.txt as {object_file}. In {object_files} all the paths are included, in {object_file} it is missing. That should not be possible at all.

We need to find out what is special about your installation. I can't reproduce any of these effects on any windows machine I try, physical or virtual machines.

Do you have something like cygwin, msys, msys2 or mingw installed on your computer?

Could you maybe start from scratch with your arduino installation? That means: Deinstall Arduino, remove (or at least rename) the AppData/Local/Arduino15 and Documents/Arduino directories in your home directory. Then reinstall Arduino and Sduino, copy the files in [wrapper1.zip](https://github.com/tenbaht/sduino/files/1440943/wrapper1.zip) into the wrapper directory to replace the normal ones? This will activate full debug output for everything. The log will be quite long. Could you send it by email to michael-mayer@gmx.de instead of filling this discussion here.

speedbug78, 2017-11-03:
>  Which filesystem are you using? FAT32?
[image]

> But why is your installation so different? The links are part of the archive and obviously the archive got unpacked. So where are the links?
That is a very good question.  Is there a way to manually unpack the download?  Having done the install through the board manager I'm not sure where the intermediate files go.  Is there a log that it keeps for installation?

> But why? This filename is generated by the Arduino IDE and used by arduino-builder with platform.txt as {object_file}. In {object_files} all the paths are included, in {object_file} it is missing. That should not be possible at all.
Another good question.  Does it have something to do with the odd characters that show up in the output and chop off my cut and paste?  Are there weird characters chopping off the build path?

> We need to find out what is special about your installation. I can't reproduce any of these effects on any windows machine I try, physical or virtual machines.
I hate problems I can't re-produce as well.  Thank you for sticking with me anyway.  It makes me wonder if somehow I've done something wrong, but the install is very simple so I can't figure out my mistake.

> Do you have something like cygwin, msys, msys2 or mingw installed on your computer?
Yes I do.  I do plenty of development work on my computer.  Here is what I have:

In Arduino itself:
The Additional Boards Manager URLs: `http://arduino.esp8266.com/stable/package_esp8266com_index.json,https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json,https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json`
As you can see I also work with esp8266 and STM32 processors

Besides Arduino I have:
System Workbench for STM32, EmBitz1.11, Bloodshed Dev-C++, RealView's ARM Workbench, Energia, MSYS2 64bit, Python2.7, MS Visual Studio 2005, and MS Visual Studio 2008.

> Could you maybe start from scratch with your arduino installation? That means: Deinstall Arduino, remove (or at least rename) the AppData/Local/Arduino15 and Documents/Arduino directories in your [...]

Yes, working on it...

speedbug78, 2017-11-03:
Here are the steps I preformed:
1. Using Control Panel->Programs and Features I uninstalled Arduino
2. Re-named C:\Users\WirickD\AppData\Local\Arduino15 to Arduino15_archive
3. Re-named C:\Users\WirickD\Documents\Arduino to Arduino_archive
4. Installed Arduino 1.8.5
5. Ran Arduino 1.8.5 (to create the Arduino15 and Arduino folders again)
6. Closed Arduino 1.8.5
7. Re-opened Arduino 1.8.5 and added "https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json" to the additional boards urls
8. Opened Board Manager and installed Sduino STM8 plan C core (non-C++) by Tenbaht
9. Extraced files from wrapper1.zip and copied them into "C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.10.21\wrapper", replacing the files that were in there.
10. Opened the Blink example, and selected STM8S103... board.
11. Pressed "Verify".  This failed with "...Couldn't find library 'STM8S103'...".  I will e-mail to you.

Should I now try with your comment below?

> As a quick fix copy the file STM8S003.lib two times to the new names STM8S103.lib and STM8S105.lib, that will fix the linking problem.

speedbug78, 2017-11-03:
I just realized that I didn't enable verbose output.  I re-verified with verbose output checked.

speedbug78, 2017-11-03:
I tried Blink with the "quick fix", per your comment and it does build to completion:

> Sketch uses 2727 bytes (33%) of program storage space. Maximum is 8192 bytes.
Global variables use 70 bytes (6%) of dynamic memory, leaving 954 bytes for local variables. Maximum is 1024 bytes.

I do not have hardware in my possession to test the build on however.

tenbaht, 2017-11-04:
> Oddly turning on verbose output in Arduino seems to have made the log
> shorter, not longer.  I've included the same build's logs one with
> verbose on and one without.

Thank you very much for the logs, they where very interesting. We are
getting closer....

The verbose log is shorter because platform.txt still contains the
"-Wall" flag instead of "-Wp-Wall". This makes SDCC failing. This is a
bug, I will correct it for the next release. Attached is a corrected
version of [platform.txt](https://github.com/tenbaht/sduino/files/1443451/platform.txt), this should correct that.

The really interesting part is the different way the source files are
referenced. Check this small part of the two logs:

### Log_without_verbose
The source files are referenced without the full path:
```text
bash.exe: warning: could not find /tmp, please create!
C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\Blink.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
+ SDCC='C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc'
+ SRC='sketch\Blink.ino.cpp'
+ OBJ='preproc\ctags_target_for_gcc_minus_e.cpp'
+ REL='preproc\ctags_target_for_gcc_minus_e.cpp.rel'
+ MARK=re12
+ shift 4
```

#### Log_with_verbose
The source files are referenced with a full absolute path:
```text
bash.exe: warning: could not find /tmp, please create!
C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc C:\Users\WirickD\AppData\Local\Temp\arduino_build_100551\sketch\Blink.ino.cpp C:\Users\WirickD\AppData\Local\Temp\arduino_build_100551\preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\cores\sduino -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\variants\standard -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0/STM8S_StdPeriph_Driver/inc -IC:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
+ SDCC='C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc'
+ SRC='C:\Users\WirickD\AppData\Local\Temp\arduino_build_100551\sketch\Blink.ino.cpp'
+ OBJ='C:\Users\WirickD\AppData\Local\Temp\arduino_build_100551\preproc\ctags_target_for_gcc_minus_e.cpp'
+ REL='C:\Users\WirickD\AppData\Local\Temp\arduino_build_100551\preproc\ctags_target_for_gcc_minus_e.cpp.rel'
+ MARK=re12
+ shift 4
```
#### Why?
This is done by arduino-builder and it has nothing to do with the used hardware package.  I can't reproduce this here using Arduino 1.8.5 on Windows7 in a virtual machine.

What was your setting for the one build that succeeded? Verbose?

What is our current state now?
- For an unknown reason the link for STM8S103.lib doesn't get generated on install. Copy the file by hand fixes that.
- platform.txt contained an error for the verbose flag. The attached file fixes that.
- does the build works now? For verbose and non-verbose setting? Or did this one build one worked because of the non-verbose compilation that you did right before and that left all needed files in the right place?

#### unpacking the archive

> That is a very good question. Is there a way to manually unpack the download? Having done the install through the board manager I'm not sure where the intermediate files go. Is there a log that it keeps for installation?

No idea about a file cache. It is downloaded from the release page: https://github.com/tenbaht/sduino/releases/download/v0.3.0/sduino-core-0.3.0.tar.bz2

Unpack it using tar with cygwin or msys: `tar xvjf archive.tar.bz2` It would be interesting to see if this generates the links correctly.

speedbug78, 2017-11-06:
I put the new platform.txt at: C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\platform.txt

> does the build works now? For verbose and non-verbose setting? Or did this one build one worked because of the non-verbose compilation that you did right before and that left all needed files in the right place?

I fired up the IDE this morning and the non-verbose build worked right off.
[arduino_build_non-verbose.zip](https://github.com/tenbaht/sduino/files/1447319/arduino_build_non-verbose.zip)

I shut down and re-started the IDE, set verbose and the build also worked.
[arduino_build_verbose.zip](https://github.com/tenbaht/sduino/files/1447318/arduino_build_verbose.zip)

[Verbose_Build_Log.txt](https://github.com/tenbaht/sduino/files/1447320/Verbose_Build_Log.txt)

I found a file cache at: C:\Users\WirickD\AppData\Local\Arduino15\staging\packages
This seems to contain the zipped packages, sdcc-stm8-i586-mingw32msvc-20171020-10088.tar.bz2, sduino-core-0.3.0.tar.bz2, and sduino-tools_mingw32-2017.10.21.tar.bz2

speedbug78, 2017-11-06:
Trying to un-tar teh sduino-core let to some interesting results:
[sduino_manual_unpack.txt](https://github.com/tenbaht/sduino/files/1447410/sduino_manual_unpack.txt)

tenbaht, 2017-11-06:
issue #25 is probably related to your problem here and I made some interesting discoveries there.

I just uploaded a new release 0.3.1 that fixes the link problem and hopefully your other difficulties as well. I discovered some subtle dependencies on a prior mingw/cygwin install and Windows environment variables that could explain all the strange effects we have seen here. Could you maybe do another fresh start with an empty Arduino and AppLocal/Arduino directory and check if it works without manual tweaks now?

speedbug78, 2017-11-06:
As a further twist, I am able to manually create the link:

`WirickD@SA-76M5VY1 MSYS ~/STM8S_StdPeriph_Driver/src
$ ln -s stm8s103.lib stm8s.lib

WirickD@SA-76M5VY1 MSYS ~/STM8S_StdPeriph_Driver/src
$ ls
doit                    stm8s_awu.c   stm8s_flash.c  stm8s_rst.c   stm8s_tim5.c   stm8s_wwdg.c
Makefile                stm8s_beep.c  stm8s_gpio.c   stm8s_spi.c   stm8s_tim6.c   stm8s103.lib
spl-splitprepare.patch  stm8s_can.c   stm8s_i2c.c    stm8s_tim1.c  stm8s_uart1.c
stm8s.lib               stm8s_clk.c   stm8s_it.h     stm8s_tim2.c  stm8s_uart2.c
stm8s_adc1.c            stm8s_conf.h  stm8s_itc.c    stm8s_tim3.c  stm8s_uart3.c
stm8s_adc2.c            stm8s_exti.c  stm8s_iwdg.c   stm8s_tim4.c  stm8s_uart4.c

WirickD@SA-76M5VY1 MSYS ~/STM8S_StdPeriph_Driver/src
$
`

speedbug78, 2017-11-06:
Ok, just to be complete I deleted the contents of my Arduino folder, and removed Arduino15.  Then I un-installed and re-installed 1.8.5.
I then added your board manager URL and installed 0.3.1
I opened Blink, and selected STM8S103F3.  I got the following warnings:
`Invalid library found in C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library: C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library
Invalid library found in C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library: C:\Users\WirickD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library
`
The build worked first time though!
I also built for STM8S105C6 and it worked too!

tenbaht, 2017-11-07:
Great! I am glad that this thing is done now!

The core problem is that the board manager unpacks the installation archives by himself instead of using a trusted external tool like tar as I expected (but never checked). That's why the links don't work out of the box. Key was the error message about the missing ln binary mentioned in issue #25.

The (unfinished) library slipped in accidentally. Ignore the message. It is only part of my development branch here. I have to work on the release scripts.

closed 2017-11-07 by tenbaht

zoomx, 2017-11-07:
Windows seven 64 bit Professional: works!

MAX6675-library is invalid, it is because sources are in a subfolder of MAX6675-library/examples

Blink take long time to compile: confirmed.

Compiling HMC5883L example: got errors
HMC5883L\HMC5883L.pde:20: error 20: Undefined identifier 'I2c'
HMC5883L\HMC5883L.pde:20: error 25: Structure/Union expected left of '.->'
HMC5883L\HMC5883L.pde:27: error 43: invalid operand for shift operator
HMC5883L\HMC5883L.pde:28: error 45: operand invalid for bitwise operation
HMC5883L\HMC5883L.pde:28: warning 113: left & right types are int fixed,void
(These errors were repeated more times on the same line instead of only one. For example the first one is repeated nine times.)
exit status 1
Error compiling for board STM8S103F3 Breakout Board.

Great Work!

MAHMOOD-SH, 2019-04-10:
hi.
I have this problem yet.
arduino 1.8.5 and 1.8.9 tested.
sduino 0.4.0.
what should I do?

Arduino: 1.8.5 (Windows 10), Board: "STM8S103F3 Breakout Board"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\MAHMOOD\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\MAHMOOD\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\MAHMOOD\Documents\Arduino\libraries -fqbn=sduino:stm8:stm8sblue -ide-version=10805 -build-path C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059 -warnings=all -build-cache C:\Users\MAHMOOD\AppData\Local\Temp\arduino_cache_453728 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.sdcc.path=C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738 -prefs=runtime.tools.STM8Tools.path=C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2018.12.07 -verbose C:\Users\MAHMOOD\AppData\Local\Temp\untitled2084226703.tmp\sketch_apr10a\sketch_apr10a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\MAHMOOD\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\MAHMOOD\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\MAHMOOD\Documents\Arduino\libraries -fqbn=sduino:stm8:stm8sblue -ide-version=10805 -build-path C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059 -warnings=all -build-cache C:\Users\MAHMOOD\AppData\Local\Temp\arduino_cache_453728 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.sdcc.path=C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738 -prefs=runtime.tools.STM8Tools.path=C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2018.12.07 -verbose C:\Users\MAHMOOD\AppData\Local\Temp\untitled2084226703.tmp\sketch_apr10a\sketch_apr10a.ino
Using board 'stm8sblue' from platform in folder: C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0
Using core 'sduino' from platform in folder: C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0
Detecting libraries used...
"C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2018.12.07/win/bash" "C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2018.12.07/wrapper/sdcc.sh" "C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738/bin/sdcc" "C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059\sketch\sketch_apr10a.ino.cpp" "nul" re12 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__  -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8   "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0\cores\sduino" "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0\variants\standard" "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0/STM8S_StdPeriph_Driver/inc" "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738/include"
Generating function prototypes...
"C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2018.12.07/win/bash" "C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2018.12.07/wrapper/sdcc.sh" "C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738/bin/sdcc" "C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059\sketch\sketch_apr10a.ino.cpp" "C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059\preproc\ctags_target_for_gcc_minus_e.cpp" re12 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__  -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8   "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0\cores\sduino" "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0\variants\standard" "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0/STM8S_StdPeriph_Driver/inc" "-IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738/include"
bash.exe: warning: could not find /tmp, please create!
C:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738/bin/sdcc C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059\sketch\sketch_apr10a.ino.cpp C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059\preproc\ctags_target_for_gcc_minus_e.cpp re12 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0\cores\sduino -IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0\variants\standard -IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.4.0/STM8S_StdPeriph_Driver/inc -IC:\Users\MAHMOOD\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10738/include
[... cut ...] 2 --iram-size 1024 -mstm8 -DSTM8S103 C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059\sketch\sketch_apr10a.ino.cpp.rel C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059/..\arduino_cache_453728\core\core_sduino_stm8_stm8sblue_e8f5dc20ce09846f4b7449265ee66d76.lib -lSTM8S103 -lstm8 --out-fmt-elf -o C:\Users\MAHMOOD\AppData\Local\Temp\arduino_build_349059/sketch_apr10a.ino.elf
ASlink-Warning-No definition of area SSEG

exit status 1
Error compiling for board STM8S103F3 Breakout Board.

NGDANGSON, 2021-08-12:
Arduino: 1.8.15 (Windows 10), Board: "STM8S103F3 Breakout Board"

C:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\Blink.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include

Mark re12:C:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\Dell M4800\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\Blink.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp

cpp gefunden

'C:\Users\Dell' is not recognized as an internal or external command,

operable program or batch file.

exit status 1

Error compiling for board STM8S103F3 Breakout Board.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

NGDANGSON, 2021-08-12:
help me