tenbaht/sduino #25: installation error on ide

beytullahgol, 2017-11-06:
can u help me ?
[image]

beytullahgol, 2017-11-06:
Cannot run program "ln" (in directory "C:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\lib"): CreateProcess error=2, Sistem belirtilen dosyayı bulamıyor
java.lang.RuntimeException: java.io.IOException: Cannot run program "ln" (in directory "C:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\lib"): CreateProcess error=2, Sistem belirtilen dosyayı bulamıyor
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Cannot run program "ln" (in directory "C:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.0\STM8S_StdPeriph_Driver\lib"): CreateProcess error=2, Sistem belirtilen dosyayı bulamıyor
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at processing.app.Platform.symlink(Platform.java:334)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:258)
	at cc.arduino.utils.ArchiveExtractor.extract(ArchiveExtractor.java:81)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:159)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: CreateProcess error=2, Sistem belirtilen dosyayı bulamıyor
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 7 more

zoomx, 2017-11-06:
Maybe there are symlinks also in Windows
[https://msdn.microsoft.com/it-it/library/windows/desktop/aa363878(v=vs.85).aspx](url)

tenbaht, 2017-11-06:
Thank you for this report, that helped a lot!
Analysing the Arduino sources I was surprised to learn that they unpack the archives themselves instead of using a standard tool like tar. This way links are not supported, even when windows can handle them now. (Finally! After only 25 years...)

Looking into this I found some other problems with the windows install. Now there is a new release 0.3.1 that should fix this. Could you please check if it works for you now and report back?

beytullahgol, 2017-11-06:
bash.exe: warning: could not find /tmp, please create!
C:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\Blink.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
Mark re12:C:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\Blink.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
bash.exe: warning: could not find /tmp, please create!
C:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\Blink.ino.cpp sketch\Blink.ino.cpp.o re2 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
Mark re2:C:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\Blink.ino.cpp -o sketch\Blink.ino.cpp.o
cpp gefunden
bash.exe: warning: could not find /tmp, please create!
--nostdlib -LC:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/lib -LC:\Users\90555\AppData\Local\Temp\arduino_build_655197 -LC:\Users\90555\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8 --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 sketch\Blink.ino.cpp.o C:\Users\90555\AppData\Local\Temp\arduino_build_655197/..\arduino_cache_370296\core\core_sduino_stm8_stm8sblue_d50ac072503b4fdcf5ca48872f7d9022.a C:\Users\90555\AppData\Local\Temp\arduino_build_655197/core/main.c.rel -lSTM8S103 -lstm8 --out-fmt-elf -o C:\Users\90555\AppData\Local\Temp\arduino_build_655197/Blink.ino.elf
Çalışmanız programın 2727 bayt (33 %) saklama alanını kullandı. Maksimum 8192 bayt.
Global değişkenler belleğin 70 byte kadarını  (6%) kullanıyor. Yerel değişkenler için 954 byte yer kalıyor. En fazla 1024 byte kullanılabilir.
Determine FLASH area

libusb: error [init_device] device '\\.\USB#VID_046D&PID_C534&MI_01#7&2FA3A65F&0&0001' is no longer connected!
libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_045E&PID_0732&MI_01#7&9C8187B&0&0001': [87] Parametre hatalı.
Writing Intel hex file 2727 bytes at 0x8000... OK
Bytes written: 2727
C:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library içerisinde geçersiz kütüphane bulundu: C:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library
C:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library içerisinde geçersiz kütüphane bulundu: C:\Users\90555\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library
[image]

ok 0.3.1 work well

tenbaht, 2017-11-07:
Great!
I see that the MAX6675 library accidentally slipped into the package. I clearly have to work on the release scripts...

closed 2017-11-07 by tenbaht