tenbaht/sduino #27: Compiling HMC5883L example: got errors

zoomx, 2017-11-07:
core 0.3.1, Windows seven 64 bit

Compiling HMC5883L example: got errors
HMC5883L\HMC5883L.pde:20: error 20: Undefined identifier 'I2c'
HMC5883L\HMC5883L.pde:20: error 25: Structure/Union expected left of '.->'
HMC5883L\HMC5883L.pde:27: error 43: invalid operand for shift operator
HMC5883L\HMC5883L.pde:28: error 45: operand invalid for bitwise operation
HMC5883L\HMC5883L.pde:28: warning 113: left & right types are int fixed,void
(These errors were repeated more times on the same line instead of only one. For example the first one is repeated nine times.)
exit status 1
Error compiling for board STM8S103F3 Breakout Board.

tenbaht, 2017-11-08:
This example is not part of Sduino and I don't find it with Arduino 1.8.5 either. Maybe you tried the ninjablocks libraries?

The error messages sound like you tried to compile unmodified C++ code. The Sduino syntax is close, but not identical to C++: https://tenbaht.github.io/sduino/api/migration/

(Brief) information about the I2C support: https://tenbaht.github.io/sduino/api/I2C/
For an example, check out the code for [Mini_SSD1306](https://tenbaht.github.io/sduino/api/Mini_SSD1306/)

Migration shouldn't be too hard. Once you got it working I would be very happy to include the example into Sduino!

tenbaht, 2017-11-08:
I wasn't aware of the way the IDE displays the examples. I am going to change the filenames accordingly to make the IDE display them correctly.

zoomx, 2017-11-08:
I am using the core installed with board manager in Windows.
I select the only STM8 breakout board and it appears as the only example available.

tenbaht, 2017-11-08:
This was a very useful hint, thank you!

Now the examples show up where they are supposed to be. I didn't port the HMC5883 example, though. The I2C library needs more work before I can finish that. I will open another issue for that and consider this issue regarding missing and miss-leading examples closed.

closed 2017-11-08 by tenbaht

zoomx, 2017-11-09:
I made a core update using the board manager and the example HMC5883 is still there. Still Windows 7 64bit Professional.
Just for information, it is a very low priority issue!

tenbaht, 2017-11-09:
yes, i know. I was debating to delete it from the repository, but I am planning to work on the i2c library this week anyway. Then I would have to re-add it again.