tenbaht/sduino #28: i2c library incomplete

tenbaht, 2017-11-08:
no support for read access yet, only write accesses are used and tested by Mini_SSD1306.

uvwxy, 2017-11-21:
My thoughts on this so far:

In `I2C.c`:` sendAddress(uint8_t i2cAddress)`:
is the third statement actually correct? 

```c
	/* Test on EV5 and clear it */
	while (!I2C_CheckEvent(I2C_EVENT_MASTER_MODE_SELECT));

	/* Send the Address + Direction */
	I2C->DR = i2cAddress;	// I2C_Send7bitAddress()

	/* Test on EV6 and clear it */
	while (!I2C_CheckEvent(I2C_EVENT_MASTER_RECEIVER_MODE_SELECTED));
```

Shouldn't we wait here for `I2C_EVENT_MASTER_TRANSMITTER_MODE_SELECTED` (in case we want to write)?

Wouldn't this mean we need two check the LSB in `i2cAddress` to determine the right event to wait for, for this to be correct?

The doc in `stm8s_i2c.h` states:
```c
    * Then the master has to wait that a slave acknowledges his address.
    * If an acknowledge is sent on the bus, one of the following events will
    * be set:
    *
    *  1) In case of Master Receiver (7-bit addressing):
    *  the I2C_EVENT_MASTER_RECEIVER_MODE_SELECTED event is set.
    *
    *  2) In case of Master Transmitter (7-bit addressing):
    *  the I2C_EVENT_MASTER_TRANSMITTER_MODE_SELECTED is set
```

Then in `receiveByte`
```c
    while (!I2C_CheckEvent(I2C_EVENT_MASTER_BYTE_RECEIVED));
     // TODO: store  I2C->DR in bytebuffer
    // TODO: send 'ack/nack'
    return 0;
```

...

uvwxy, 2017-11-23:
You can assign this one to me. I just read a single byte from I2C..

uvwxy, 2017-11-23:
I started a complete rewrite of I2C.c / I2C.h, as most of the functionality is handled by stm8s_i2c anyway...

[It's work in progress](https://github.com/uvwxy/sduino/tree/test/I2C)

I am reading the data fields (8 bytes) from a BME280 sensor and print the bytes to the OLED (poor man's debugging)...

tenbaht, 2017-11-23:
Sounds good, thank you!

stm8s_i2c is part of the SPL. It is used sometimes for historic reasons, but my goal is to remove all dependencies and include it only as an optional library that could be used, if that seems to be useful for a project (because it might be ported to STM32 later).

For the core files (wiring.c, wiring_analog.c, HardwareSerial.c) I use a compiler flag USE_SPL to separate the parts requiring SPL and the stand-alone parts from each other. I think it would be helpful if you do something similar for i2c.

In general most spl function just test for one flag or do a simple register access. In most cases that is not really useful and can easily be replaced. But there are some exceptions. The uart initialization is a good example. It does useful calculations and it couldn't be done much simpler, so it might be more useful to avoid duplicating code and just stay with the spl function.

tenbaht, 2017-11-23:
I tried to assign this one to you but it doesn't accept your username. Don't know why.

uvwxy, 2017-11-24:
Good point. I will introduce the USE_SPL flag into the I2C library.

uvwxy, 2017-11-24:
> I tried to assign this one to you but it doesn't accept your username. Don't know why.

uvwxy ? awww 😔

>For the core files (wiring.c, wiring_analog.c, HardwareSerial.c) I use a compiler flag USE_SPL to separate the parts requiring SPL and the stand-alone parts from each other. I think it would be helpful if you do something similar for i2c.

[Am I doing this right?](https://github.com/uvwxy/sduino/commits/test/I2C)

I'm not considering this as stable, but it works for me 🤓

uvwxy, 2017-12-02:
I added an example for ADXL345, [here](https://github.com/uvwxy/sduino/tree/test/I2C).

Unfortunately I have not been very successful with the I2C on the stm8s103.  I can read a few bytes here and but it is way too buggy to be useful. Also, writing to I2C crashes here and there so I'm not sure if I made things worse on my branch. 😅

closed 2018-11-14 by tenbaht

aryeguetta, 2018-11-15:
Hi, I am new and I would like to try STM8S boards as well - already got me a couple to start with - the basic cheap ones - 2 under 1 USD :)
I have set up the JSON and installed the boards from the board manager - I have also tried the manual way but with no luck at all.
I have issues that every time I upload a sketch I get a warning about /tmp folder that needs to be created for the bash - I do have it and at the end, it uses it - I wonder why I am getting this error.

I would be happy to learn more and help if needed :) 
I am using Windows 8.1 and already have my ST-LINKV2 connected and working
I was able to upload only the blink sketch and I do want to be able to do more.

Thanks

commit e429fc04 (2018-11-14): fix #28: I2C library complete, updated documentation