tenbaht/sduino #29: sduino not support mac OSX

artchula, 2017-11-09:
I was install sduino dialog box recipe message ‘Tool STM8Tool is not available for your operation system’.

dubelectro, 2017-11-10:
I was able to manually install and run it under macOS. I may be able to help.

dubelectro, 2017-11-10:
Since I am new to Github, I uploaded a set of files for macOS to Google Drive.
https://drive.google.com/open?id=1UBRyeSI1V-WN8ZisNwVY_2OwUAVVoMDq

Please unzip it under "/Users/<username>/Documents/Arduino/hardware".
Then start up Arduino IDE and you will be able to use Sduino.

The differences from the original are below
- Added sdcc for macOS(sduino/tools/sdcc/macosx).
- Added stm8flash binary compiled by myself(sduino/tools/macosx).
- Modified two locations in platform.txt for macOS. The difference is below.
===========
--- platform_original.txt 2017-11-09 06:05:34.000000000 +0900
+++ platform_macosx.txt 2017-11-10 16:06:31.000000000 +0900
@@ -14,6 +14,7 @@
 # For Board Manager installs these variables are already predefined and will
 # not be overwritten. This assignment is done only for manual installs.
 runtime.tools.sdcc.path={runtime.hardware.path}/tools/sdcc
+runtime.tools.sdcc.path.macosx={runtime.hardware.path}/tools/sdcc/macosx
 runtime.tools.STM8Tools.path={runtime.hardware.path}/tools
 
 
@@ -127,7 +128,7 @@
 tools.stm8flash.cmd=stm8flash
 tools.stm8flash.path={runtime.tools.STM8Tools.path}/linux
 tools.stm8flash.path.windows={runtime.tools.STM8Tools.path}/win
-tools.stm8flash.path.macos={runtime.tools.STM8Tools.path}/macosx
+tools.stm8flash.path.macosx={runtime.hardware.path}/tools/macosx
 #tools.stm8flash.path.linux64={runtime.hardware.path}/tools/linux64
 
 #tools.stm8flash.upload.params.verify=
===============

tenbaht, 2017-11-10:
Thank for this test!
I see that you are using the older sdcc snapshot build 9960. The reason that kept me from integrating that into the regular release is that this is different from the versions for Linux and Windows and the board manager files are not flexible enough to specify different tool versions for different platforms. I would have to upload it as a fake version 10088 - not beautiful, but possible.

About platform.txt:
If you move the sdcc files from tools/sdcc/macos one level up into tools/sdcc it should work without the first modification.
And are you sure the second modification is needed? Could you please try without? I would expect it to come out fine both ways.
If it works, than I will include macos files for the 0.3.3 release (coming soon)

dubelectro, 2017-11-11:
The snapshot of the published mac version seems to be old. I was able to build SDCC from the latest source with macOS, so I can provide it if needed. However, I think that there is no problem with your way.

**platform.txt**
> first modification

I agree. I did so because I wanted to coexist with other platforms by manual installation, but I think that there is no problem if the specification of the path is correct. 

> second modification

I do not know the cause, but it seems not to work without doing this.
Java error will be displayed.
```
Java.io.IOException: Cannot run program "{runtime.tools.STM8Tools.path}/macosx/stm8flash": error=2, No such file or directory
```

It seems that it can not find a real path if assignment nests. It is probably a macOS specific issue.

In addition, `dash` is not installed by default on Mac, but seems to work without problems.

tenbaht, 2017-11-11:
Oh, yes, I would love to add a recent sdcc binary for Mac. For Linux and Windows I used the snapshot 10088. Would it be possible that you compile the same revision for Mac? It is not strictly required to use the same version across systems, but I am afraid to run into hard-to-trace problems between different installations.

This platfom.txt thing is weird. To check if it is related to nested assignments: Could you maybe change the last line of platfom.txt to use {runtime.tools.STM8Tools.path} directly:

```text
tools.stm8flash.upload.pattern="{runtime.tools.STM8Tools.path}/macos/{cmd}" -c{upload.protocol} -p{upload.mcu} -s flash -w "{build.path}/{build.project_name}.hex"
```

tenbaht, 2017-11-11:
Ahh! It was right there and we didn't see it!

The second modifiation: I used "path.macos" instead of "path.macosx"

I bet we don't need the first modification anymore and this works:
```diff
@@ -127,7 +128,7 @@
 tools.stm8flash.cmd=stm8flash
 tools.stm8flash.path={runtime.tools.STM8Tools.path}/linux
 tools.stm8flash.path.windows={runtime.tools.STM8Tools.path}/win
-tools.stm8flash.path.macos={runtime.tools.STM8Tools.path}/macosx
+tools.stm8flash.path.macosx={runtime.tools.STM8Tools.path}/macosx
 #tools.stm8flash.path.linux64={runtime.hardware.path}/tools/linux64
```
Could you please check this?

dubelectro, 2017-11-11:
I was aware that it was "macos". I changed that, but it did not improve.

`tools.stm8flash.upload.pattern="{runtime.tools.STM8Tools.path}/macosx/{cmd}" -c{upload.protocol} -p{upload.mcu} -s flash -w "{build.path}/{build.project_name}.hex"`

This also causes an error. The error message is as follows.

```
"/Users/dub/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-objcopy" -O ihex -R DATA -R INITIALIZED -R SSEG  "/var/folders/pz/78z11q854g13g9wxpj1_zpkh0000gn/T/arduino_build_463351/Blink.ino.elf" "/var/folders/pz/78z11q854g13g9wxpj1_zpkh0000gn/T/arduino_build_463351/Blink.ino.hex"
Sketch uses 2759 bytes (33%) of program storage space. Maximum is 8192 bytes.
Global variables use 70 bytes (6%) of dynamic memory, leaving 954 bytes for local variables. Maximum is 1024 bytes.
{runtime.tools.STM8Tools.path}/macosx/stm8flash -cstlinkv2 -pstm8s103?3 -s flash -w /var/folders/pz/78z11q854g13g9wxpj1_zpkh0000gn/T/arduino_build_463351/Blink.ino.hex 
java.io.IOException: Cannot run program "{runtime.tools.STM8Tools.path}/macosx/stm8flash": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2178)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 10 more
An error occurred while uploading the sketch
```

`tools.stm8flash.upload.pattern="{runtime.hardware.path}/tools/macosx/{cmd}" -c{upload.protocol} -p{upload.mcu} -s flash -w "{build.path}/{build.project_name}.hex"`

This is ok.

Please wait for SDCC build 10088 a little more.

tenbaht, 2017-11-12:
Strange.
Unfortunately, {runtime.hardware.path}/tools directly wouldn't work for automatic installs using the board manager. The board manager adds two more directory level before tools: STM8Tools/2017.11.06/tools. The second one depends on the version string given in the package_index.json file and is different all the time. It would require really ugly tricks to work around this.

Did you try my earlier suggestion based on "{runtime.tools.STM8Tools.path}/macos/{cmd}" ? This would be a useful intermediate solution.

dubelectro, 2017-11-12:
`"{runtime.tools.STM8Tools.path}/macos/{cmd}"`
This was also the same result. The same error.

Although it is not a smart way, in macOS the folder where the board is installed is fixed, so it may be certain to specify it with full path. Every time you have to rewrite the installation script according to the version number. It may be a little hard.

tenbaht, 2017-11-12:
Yes, that would the last resort. But it works fine for all the other variables and it works fine for the other platforms and I don't understand what is different for this one case.

Today I tried to set up osx with virtual box to look into this, but it didn't work out so far as it is pretty hard to find a useful installation image.

It looks like runtime.tools.STM8Tools.path works when used for compiler.wrapper.path, but not for tools.stm8flash.path.macosx. So what is different? The .macosx specifier at the end? What happens if you use this instead?

```text
tools.stm8flash.cmd=stm8flash
tools.stm8flash.path.linux={runtime.tools.STM8Tools.path}/linux
tools.stm8flash.path.windows={runtime.tools.STM8Tools.path}/win
tools.stm8flash.path={runtime.tools.STM8Tools.path}/macosx
```
Or even further, just as a test:
```text
tools.stm8flash.cmd=stm8flash
tools.stm8flash.path={runtime.tools.STM8Tools.path}/macosx
```

dubelectro, 2017-11-13:
Both of them were same errors.
However, I solved a secret a little.

{runtime.tools.STM8Tools.path} behaves like predefined property when installing from the board manager. For example, it shows the path as below.
`/Users/dub/Library/Arduino15/packages/sduino/tools/STM8Tools/0.3.2/`
This is the path including the version number.

In the case of manual installation, {runtime.tools.STM8Tools.path} does not act as Predefined property. Therefore, it did not work in my environment.

In your `platform.txt`, it works because it defines {runtime.tools.STM8Tools.path} in the first block. In the case of macOS, I assume that it got an error trying to act as the Predefined property only for the last block.  It is probably a bug in Arduino IDE.

In other words, when installing from the board manager, you do not need to define {runtime.tools.STM8Tools.path} in the first block. It works even without it.
And
`tools.stm8flash.path.macosx={runtime.tools.STM8Tools.path}/macosx`
 line seems to be work as it is.

I noticed these differences compared to …
https://github.com/stm32duino/Arduino_Core_STM32
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/platform.txt
I tried installing sduino pretending to be Arduino_Core_STM 32,  it worked. I changed part of `plarform.txt` according to Arduino_Core_STM 32, so it is not a perfect test.

Would you please publish the test installation script if possible? If I use it, I can test more certainly.

Next is about sdcc.
I tried building various revisions from sources. Build can be done without problem. But revision 10088 does not work with sduino.
Revision 10174, 9960 is OK。The revision between 9960 and 10174 (although I have not tried everything) was almost bad.
I guess that it is why sndshots for macOS 9960 and later are not published on sdcc's site. For the above reasons, it seems to be only possible to use 9960 or 10174 for macOS.

tenbaht, 2017-11-13:
> In your platform.txt, it works because it defines {runtime.tools.STM8Tools.path} in the first block. In the case of macOS, I assume that it got an error trying to act as the Predefined property only for the last block.
>  It is probably a bug in Arduino IDE.

Difference is that the last block is evaluated by the Arduino code itself (written in java), the former blocks are evaluated by arduino-builder (written in go). They don't share any code, so it might be possible that things are slightly different between the two. But investigating the sources in arduino-core/src/processing/app/helpers/PreferencesMap.java I can't find anything suspicious.

Only thing I see in arduino-core/src/processing/app/BaseNoGui.java:184 is the handling of the requiredTools for a platform. But if somehow the system is still remembering some data from a former failed board manager installation attempt, we should see a (invalid) path replacement for runtime.tools.STM8Tools.path.

But we don't see anything being replaced. And the definition in the first block is completely ignored.

I am not sure what this means:
> In other words, when installing from the board manager, you do not need to define {runtime.tools.STM8Tools.path} in the first block. It works even without it.
> And tools.stm8flash.path.macosx={runtime.tools.STM8Tools.path}/macosx
line seems to be work as it is.

> I tried installing sduino pretending to be Arduino_Core_STM 32, it worked. I changed part of plarform.txt according to Arduino_Core_STM 32, so it is not a perfect test.

What exactly works for you? And what doesn't? It sounds like for the board manager install runtime.tools.STM8Tools.path gets replaced, but for the manual install it is not possible to set it to a useful value. Is this right?

> Would you please publish the test installation script if possible? If I use it, I can test more certainly.

The board manager information is in [package_sduino_stm8_index.json](https://github.com/tenbaht/sduino/blob/master/package_sduino_stm8_index.json). You can modify that file on your computer and use a URL like `file:///path/to/my/package_index.json` for the board manager for local tests. Add your sdcc package into the sdcc section at the end and you can install it using the board manager. Just for testing you could even just copy the sdcc-for-linux block and edit it to `"host": "x86_64-apple-darwin"`. It will install the linux binaries, but you can replace these files afterwards by hand with your sdcc version without breaking anything.

It would be great to solve this mystery of the failing property setting.

dubelectro, 2017-11-13:
>  It sounds like for the board manager install runtime.tools.STM8Tools.path gets replaced, but for the manual install it is not possible to set it to a useful value. Is this right?

Yes. I think so.
I modified package_sduino_stm8_index.json by myself and installed from the board manager.
It seems that it could be installed without problems. Then I changed the following.

- Replaced with sdcc for macOS
- Added stm8flash for macOS.
- Modified `macos` of last block of `platform.txt` to `macosx`

Then all worked fine! There seems to be no problem with both build and write to the board.

At first, I tried it by manual installation, this is a different place than installing with board manager. 
I confirmed that the handling of `runtime.tools.STM8Tools.path` is different between the case of installing with board manager and manual installation, but I do not know the reason.

tenbaht, 2017-11-13:
Good! So I am going to prepare 0.3.3 with proper MacOS support now.

Should I integrate your sdcc version? So far the only thing I have is the old 9960 snapshot build. You could attach your build here. (please leave out the PIC-related directories, they are so big)

dubelectro, 2017-11-13:
I uploaded sdcc 10174 for macOS.
https://drive.google.com/open?id=1wYvpT0VLVTmy9lVPZrd_SnZEt0Fdq9es
I built this with targeting only STM8, so there should not be unnecessary files.
I think that there is no problem as far as I tried, but if there is any problem please use 9960.

closed 2017-11-13 by tenbaht

tenbaht, 2017-11-14:
Thank you very much for that work! That's amazing, now we can have full support for all three platforms!

I was tempted to use the 10174 right away, but that would be to big a change for just a bugfix release. This is asking for 0.4.0 and I would like to test a little bit more before using it because there was one important change in code generation.

For 0.4.0 I am planning to add support for the builtin bootloader of the STM8S105 using [stm8gal](https://github.com/gicking/stm8gal). Have you used that program before? If yes, it would be great if you could send me a macosx binary that I could include with the release.

Again, thank you for your help here!

dubelectro, 2017-11-14:
Thank you for your hard work.

I tried a new version 0.3.3.
It was installed without any problem.
However, there was one mistake left.
 `platform.txt` in `sduino-core-0.3.3.tar.bz2` has not been updated.
`tools.stm8flash.path.macos = {runtime.hardware.path}/tools/macosx`
It remained on. (macos -> macosx)
The repository seems to have no problem.

I have not tried stm8gal yet.
I am ordering a new STM8S105 board, so I will try it when I receive it.
I could build stm8gal so I could provide it.

tenbaht, 2017-11-14:
> However, there was one mistake left.
platform.txt in sduino-core-0.3.3.tar.bz2 has not been updated.
tools.stm8flash.path.macos = {runtime.hardware.path}/tools/macosx

Damn it :-( Yeah, now I see how that happened. The release script carefully packages the latest committed versions of the files - and I committed the platfom.txt change *after* generating the core archive file.

As an exception I violated the rule of never changing a release by uploading the correct core archive now. At least it now really contains the files tagged with v0.3.3 in the repository.

commit 2cf64142 (2017-11-13): fix #29: add support for board manager install on Mac OSX