tenbaht/sduino #31: How to select Port in Windows ?

albertlu2013, 2017-11-13:
I can not find the Port for the stm8s105c6t6 discovery in the Arduino IDE.
How to select Port for it in Win7 ?

[image]

tenbaht, 2017-11-13:
The upload is done via the builtin ST-Link interface using stm8flash. That means it doesn't use a serial port and there is no need to select any, but you need to download and install the ST-Link/v2 driver **before you plug in the flash tool**: 
[STSW-LINK009 driver download](http://www.st.com/en/development-tools/stsw-link009.html)
(Registration required, but very easy)

closed 2017-11-13 by tenbaht

albertlu2013, 2017-11-15:
Hi Michael Mayer

  Thanks for your reply .
  I have installed the STSW-Link009 driver .
  I get the error message :

Determine FLASH area

libusb: info [cache_config_descriptors] could not access configuration
descriptor (dummy) for '\\.\USB#VID_0A5C&PID_21E6#B8763FAE915E': [31]
連結到系統的某個裝置失去作用。
libusb: info [cache_config_descriptors] could not access configuration
descriptor (dummy) for '\\.\USB#VID_147E&PID_2020#6&30868024&0&3': [31]
連結到系統的某個裝置失去作用。
Could not open USB device.
An error occurred while uploading the sketch

Do you know How can I to do ?

Thanks

ᐧ

commit 9b107225 (2017-11-13): fix #31: pointing out the need to install a driver for ST-Link