tenbaht/sduino #32: [Help] VirtualWire for STM8

Jugulaire, 2017-11-16:
Hi,

I'm working on project who need 433Mhz RF communication. 
I would like to use STM8 instead of ATMEGA for doing it.

I've read how to convert Arduino Lib to STM8 Lib but i'm not sure how to do this. 
Is anybody already have convert VirtualWire to STM8 ? 
Is anybody know how i can do this ? 

Thanks for helping !

tenbaht, 2017-11-16:
Oh, that sounds like in interesting challange!
- The lib uses a C interface instead of a C++ class, thats good.
- It already deals with different platforms, so we might be able to integrate STM8 with the existing code. Good.
- It works very close to the hardware: Timer and ISR programming. That code needs to be re-written from scratch and carefully woven into the existing Sduino code. A lot of debugging work waiting here.
- The main file is a cpp file containing almost pure c code. Simple renaming (as I did with the other libs) would be possible, but it would break the option to really integrate it with the existing code. So it might require some extensions  of the build scripts. But that has to be done anyway at one point. Not too hard.

It will be one of the harder libs to convert. Definitely not a one-evening-thing. It would be great to see this working. I even a couple of these 433MHz modules here in my desk waiting to be used for experiments.

I would love to work on this, but it might take a longer time before I would be able to find the time to do so - so don't hold your breath. Fixing i2c lib and integrating bootloader handling are first on my list.

tenbaht, 2017-11-16:
I converted Mike McCauley's original zip archive into a proper github repository, including full version history: https://github.com/tenbaht/VirtualWire
Maybe that is a useful good base for developments.

Jugulaire, 2019-04-08:
I think it can be more easy to adapt the manchester library for doing this : 
https://github.com/cano64/ManchesterRF

I would like to try to work on adaptation.. good way to learn

mbta009, 2019-09-29:
Hi, I'm working on Similar project, 433 RF with STM8s003F3p6 
Did you come up with a solution to convert the library?
Thanks for sharing the source code for STM8.

Dhiraj161993, 2022-01-02:
Hi, I'm working on a project, 433 RF with STM8s003F3p6  can u share with me the process, how can I receive data and control the relay

Jugulaire, 2022-01-31:
Hi, i have tried to adapt the 443 MHz library to the STM8, and it has never worked correctly. 

As explained before, i'm still struggling into the Manchester code adaptation.