tenbaht/sduino #33: I2C.c: Undefined identifier 'TWDR'

uvwxy, 2017-11-18:
First of all, great job putting all this together. I'm using sduino via makefile include on macOs. So far I got the blink example to work using the stlink v2 😀👍

My next step is to get the Mini SSD1307 example to work. Currently I get the following compile error:

```bash
[...]/sduino/hardware/sduino/stm8/libraries/I2C/I2C.c:398: error 20: Undefined identifier 'TWDR'
make: *** [build-stm8sblue/platformlibs/I2C/I2C.c.rel] Error 1
```

The code at the location is the following:

```c
uint8_t I2C_read(uint8_t address, uint8_t numberBytes)
{
  bytesAvailable = 0;
  bufferIndex = 0;
  if(numberBytes == 0){numberBytes++;}
  nack = numberBytes - 1;
  returnStatus = 0;
  returnStatus = start();
  if(returnStatus){return(returnStatus);}
  returnStatus = sendAddress(SLA_R(address));
  if(returnStatus)
  {
    if(returnStatus == 1){return(5);}
    return(returnStatus);
  }
  for(uint8_t i = 0; i < numberBytes; i++)
  {
    if( i == nack )
    {
      returnStatus = receiveByte(0);
      if(returnStatus == 1){return(6);}

      if(returnStatus != MR_DATA_NACK){return(returnStatus);}
    }
    else
    {
      returnStatus = receiveByte(1);
      if(returnStatus == 1){return(6);}
      if(returnStatus != MR_DATA_ACK){return(returnStatus);}
    }
    data[i] = TWDR;
    bytesAvailable = i+1;
    totalBytes = i+1;
  }
  returnStatus = stop();
  if(returnStatus)
  {
    if(returnStatus == 1){return(7);}
    return(returnStatus);
  }
  return(returnStatus);
}
```

Should the line with `data[i]` be set to `address`?

Hint: I have used esp8266's with 128x64 oled displays with I2C/SPI successfully so far, but have no clue about the details of I2C.

uvwxy, 2017-11-18:
I set `data[i] = address` just to get past the compiler error ... and the example works.

Closing, as we already have this ticket: #28

closed 2017-11-18 by uvwxy

aryeguetta, 2018-11-15:
I am trying this example with my STM8S103F board and I am getting:
`Global variables use 1141 bytes (111%) of dynamic memory, leaving -117 bytes for local variables. Maximum is 1024 bytes.`
What should I do to make it works?
Thanks

commit 8638b9c7 (2017-12-16): fix #33: comment out function i2c_read() as it is not ported yet