tenbaht/sduino #38: How to program the chips of stm8s103f?

thoulin, 2017-12-20:
In windows envirement, the sduino core can be merge into Arduino IDE (version of 1.6.8),  I cannot upload the program into chips,can you point out my fault? I have stlink v2 dongle & USB to series tool kit.  I cannot   install the driver for sduino.

tenbaht, 2017-12-20:
For Windows, you need to install the ST-Link driver from the ST website first.
See https://tenbaht.github.io/sduino/hardware/flashtool/
Hope that helps.

thoulin, 2017-12-21:
I have installed the STLINK V2 DRIVER，but in arduino envirment, it can not be found. and can you point out more detail of SDCC's installing? and how to link with stlink v2 hardware?

tenbaht, 2017-12-21:
I am afraid I don't understand what exactly is not working.
Did you do a normal install using the board manager? This will include sdcc, stm8flash and stm8gal, no more manual installations are required.
Could you maybe copy the error message that you get or attach a screenshot from the point where you are stuck?

MrAmazoni, 2018-04-08:
I have the same problem.
[image]

zoomx, 2018-04-10:
The STlink drver has also a small utility: STM32 ST-LINK Utility
Check that this utility recognizes the chip. If not there is a communication issue.
Take in account that some STlink clones doesn't work with STM8 but only with STM32. I had one of them.

closed 2018-10-08 by tenbaht