tenbaht/sduino #40: RF24 port

mcer12, 2017-12-29:
Hi @tenbaht or anyone who might be interested in porting this library. Since I'm not skilled/experienced enough to archieve this on my own, I thought it might be worth putting it here. With this library working, the stm8 can become real IoT platform.

I found a fork already ported to c if that's of any help.
https://github.com/nRF24/RF24/issues/328

As I don't know it this is even possible, I put this here for a possible further discussion.

jsadv, 2018-01-07:
I have been trying to locate an appropriate src c code that would allow me to run c code in arduino C++.  Mostly alot of compile errors as most src files are IAR tuned. I am hoping some can make this possible. I would like to transfer many protocols stm8 via sduino, serial,midi etc..

I`m still hoping :)

tenbaht, 2018-01-07:
Sounds very interesting to me. I even have some of these modules laying around for some years already waiting for experiments. But I won't have enough time in the next few months to look into it more deeply. Maybe somebody else could try it?

mcer12, 2018-01-07:
Thanks for the reply @tenbaht , if you know of anyone who might be interested please tag them.

jsadv, 2018-01-24:
Got this working STM8s and nrf24l01+i. Its all GNU code .. right ? how do i share it. I am new to this .
I have been able to transfer 8 bit integers via the 4 byte array using nrf24 . Sduino rocks

/* jsadv@yahoo.com ..I have put this code to gether from bits and pieces 
 * I am a novice programmer big on learning.. cheers
 * borrowed code from   Ishan Kehribar (Little Wire…) submitted the following code for nrf24l01 :
 *  https://github.com/kehribar/nrf24L01_plus .
 * Modified  stuff etc
 * http://blog.pommerie-michel.fr/?p=179
 * I am sure someone with more experience can to better,
 * but it works for me I can control anything with this simple SDUINO app
 * I am sure the code can be made cleaner and faster. I just wanted it to work :) and it works great!!!
 * Thanks tenbaht */
[wifib.tar.gz](https://github.com/tenbaht/sduino/files/1658154/wifib.tar.gz)
[wifia.tar.gz](https://github.com/tenbaht/sduino/files/1658155/wifia.tar.gz)

jsadv, 2018-01-24:
Works also when I transfer left adjusted ADC  8 bit. Sorry about the extra code in file , forgot to clean it up. Has lots of redundant stuff.

tenbaht, 2018-01-24:
Great, thanks! It looks like you developed it as a stand-alone project, not as a full sduino library, right? But the code looks like it wouldn't be hard to split it into a library part and some example code. Maybe you could have a look into that?
The cleanest way to get code integrated would be a fork and a pull request. That means you fork sduino into your account, clone that repository to your harddrive, integrate your work as a library and send me a pull request when you are satisfied with it.
I would love to integrate it with sduino, but I am traveling now and won't work on it for the next few weeks. I keep it in mind!

jsadv, 2018-01-25:
i await there expertise of some other...noted....

anshulteotia, 2018-02-09:
i am working with stm8+nrf24l01, can you please provide me with the codes

Faqarini, 2019-12-07:
Hi, I would like to make a RC receiver with a STM8s and a nrf24l01 module. Is there a way to make it possible with Sduino?

TurtleForGaming, 2020-01-26:
Is there an update on porting this library ?

stefaandesmet2003, 2021-02-12:
For what it's worth, I have a working light version of MySensors on my github repo, using the RF24 implementation that comes with MySensors. The example is a simple node sending a blinkie to a MySensors gateway.
It's a work in progress however. But feel free to use and test.

https://github.com/stefaandesmet2003/sduino/tree/master/hardware/stm8/0.5.0/libraries/MySensorsLight