tenbaht/sduino #41: How to write library to sduino

Ikity, 2018-01-24:
I was looking a way to port tm1637, but im not skilled in creating libraries,  so i looked in tutorial of creating own libraries on arduino official cite, and there was simple morse library with prepared header and cpp file but for some reason i wasnt able to compile anything with it.Could you give me simple example of library,  so i tried on my own

tenbaht, 2018-01-24:
Oh, yes, that is on my would-like-to-do-list as well. Two different modules with these chips are already waiting for me on my desk. But I won't be able to find the time to do so in the next few months...
There is no need to start from scratch. You could use this as a starting point: https://github.com/avishorp/TM1637
There is some brief information about the preprocessor tricks used for sduino on the website. To keep things simple you build it a single instance library with constructor, as it is done for LiquidCrystal:
https://tenbaht.github.io/sduino/developer/macro/#single-instance-with-constructor
Hope that helps

Ikity, 2018-01-24:
In your liquidCrystal library variant int header you start with something like 

```
void LiquidCrystal_init(uint8_t fourbitmode, uint8_t rs, uint8_t rw, uint8_t enable,
	    uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3,
	    uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7);
```

I tried like this for at least some beginning. 
```
#ifndef __TM1637DISPLAY__
#define __TM1637DISPLAY__
#include <xmacro.h>
//#include <inttypes.h>

#define SEG_A   0b00000001
#define SEG_B   0b00000010
#define SEG_C   0b00000100
#define SEG_D   0b00001000
#define SEG_E   0b00010000
#define SEG_F   0b00100000
#define SEG_G   0b01000000

#ifdef __cplusplus

// C++ interface for use with regular gcc compiler

#else
void TM1637Display(uint8_t pinClk, uint8_t pinDIO);

#endif

#endif // __TM1637DISPLAY__

```

```
#include <TM1637Display.h>
#include <Arduino.h>

#define TM1637_I2C_COMM1    0x40
#define TM1637_I2C_COMM2    0xC0
#define TM1637_I2C_COMM3    0x80

void TM1637Display(uint8_t pinClk, uint8_t pinDIO)
{
	// Copy the pin numbers
	m_pinClk = pinClk;
	m_pinDIO = pinDIO;

	// Set the pin direction and default value.
	// Both pins are set as inputs, allowing the pull-up resistors to pull them up
    pinMode(m_pinClk, INPUT);
    pinMode(m_pinDIO,INPUT);
	digitalWrite(m_pinClk, LOW);
	digitalWrite(m_pinDIO, LOW);
}

```
and well I've got this kind of error

> Arduino/libraries/TM1637-master/TM1637Display.h:36: syntax error: token -> 'pinClk' ; column 33
how do you make this init

Ikity, 2018-01-27:
I think it works can you check it https://github.com/Ikity/tm1637.git

closed 2018-01-31 by Ikity