tenbaht/sduino #46: Mini SSD1306 not compiling :(

regimantas, 2018-02-04:
Arduino ide version 1.6.13

Output:
bash.exe: warning: could not find /tmp, please create!
C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\oled-mini.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Mini_SSD1306 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
Mark re12:C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Mini_SSD1306 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\oled-mini.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
bash.exe: warning: could not find /tmp, please create!
C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\oled-mini.ino.cpp sketch\oled-mini.ino.cpp.o re2 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Mini_SSD1306 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
Mark re2:C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Mini_SSD1306 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\oled-mini.ino.cpp -o sketch\oled-mini.ino.cpp.o
cpp gefunden
bash.exe: warning: could not find /tmp, please create!
C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C\I2C.c libraries\I2C\I2C.c.o re1 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Mini_SSD1306 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
Mark re1:C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Mini_SSD1306 -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C\I2C.c -o libraries\I2C\I2C.c.o
C:\Users\Avitela\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\I2C\I2C.c:398: error 20: Undefined identifier 'TWDR'

exit status 1
Error compiling for board STM8S103F3 Breakout Board.

uvwxy, 2018-02-11:
Hi, I had the same problem. If you don't mind testing things you can try my branch. I will create a pull request for this in the future.

https://github.com/uvwxy/sduino/commits/requests/i2c

tenbaht, 2018-11-14:
I reworked and completed the I2C lib, commit e429fc04 should fix all this.

closed 2018-11-14 by tenbaht