tenbaht/sduino #48: PWM issue > analogWrite(LED_PIN2,255);  and  analogWrite(LED_PIN2,0);

banana2018a, 2018-02-28:
I am trying to do the LED blinking by adjust speed and brightness via A0 and A1.
I have some problem when try to use analogWrite(LED_PIN2,255);  and  analogWrite(LED_PIN2,0);
The LED is not working by fading from 0 to 255.  It happended that the LED blink randomly. 
But when change to  analogWrite(LED_PIN2,254);  and  analogWrite(LED_PIN2,1);  it is work.
My code is here:-

// ===================
  int DELAYvalue;
 int BRIGHTvalue;
   int ia=0;
#define LED_PIN1      13  // PD4
#define LED_PIN2      PD3
//#define LED_PIN      PA3  // PD4
//#define LED_PIN      PC7
#define SPEEDADC_PIN  A1
#define BRIGHTADC_PIN  A0
void BLINKLED(void);
 
void setup() {
      Serial_begin(115200);
      pinMode(SPEEDADC_PIN , INPUT); 
      pinMode(LED_PIN1,OUTPUT); 
       pinMode(LED_PIN2,OUTPUT); 
}

 
void loop() {
     ia++;

     while(1)
     {
      ia++;
      if(ia>1024) { ia=0;}
      BLINKLED();
     }
  
}

void BLINKLED(void){
      BRIGHTvalue = analogRead(BRIGHTADC_PIN);
      BRIGHTvalue =    BRIGHTvalue/4;
     DELAYvalue = analogRead(SPEEDADC_PIN); 
  
     Serial_print_s("BRIGHT = ");  Serial_println_u(BRIGHTvalue);  
      analogWrite(LED_PIN1,BRIGHTvalue);   // turn the LED on 
       analogWrite(LED_PIN2,BRIGHTvalue); 
        delay(DELAYvalue);          

      analogWrite(LED_PIN1,255);    // turn the LED off by making the voltage HIGH ( Active Low)
      analogWrite(LED_PIN2,255);   
      delay(DELAYvalue); 
  
 
}

csnol, 2018-08-25:
@banana2018a 

You can try that:

``` C++
// for analogWrite(pin, 0)
analogWrite(LED_PIN1,1);
delay(2);
digitalWrite(LED_PIN1, 0);

// for analogWrite(pin, 255)
analogWrite(LED_PIN1,254);
delay(2);
digitalWrite(LED_PIN1, 1);

```

tenbaht, 2019-01-08:
This sounds a lot like #72. This should be fixed by pr #73 now.

closed 2019-01-08 by tenbaht