tenbaht/sduino #50: Any possibility of adding the EEPROM library to this project?

ElDominio, 2018-03-14:
I've been a good two or three hours figuring out how to get started with SDCC, and finally got it to work after some mashing of my head against the wall.
How difficult does this library look to be able to port it to sduino? I tried comparing the I2C library from Arduino vs the one in sduino and I understood pretty much nothing of how the conversion was done.
Thanks for the great work you've already done though, tenbaht!

tenbaht, 2018-10-08:
The I2C lib is a mess and needs a rewrite, on my todo already. But EEPROM clearly needs to be ported. Your request is very old, but now in the fall I hope to find time for private projects again. Our little guy is finally starting at daycare now and that hopefully helps to free up some time.

ElDominio, 2018-10-12:
Heh, better late than never! If you need any type of testing, let me know!

And did not know you were a father, family always comes first!

ElDominio, 2018-12-03:
Gonna try it out this week at the latest! Thank you sir!

tenbaht, 2018-12-05:
Board manager works only for release versions. I am working on 0.4.0 and hope to finish that this week, but we will see. In the meantime, you can try to copy the files from the development branch into your existing install. Not sure if it will really work, as this will still use the older compiler version.