tenbaht/sduino #51: OSX: Arduino IDE verify fails

KenjutsuGH, 2018-03-23:
Hi,

Using Arduino 1.8.5 under OSX 10.12.6 I can use **File -> New** to create a new .ino file. I can choose either of the STM8 board, hit **Verify**, and it compiles. However, as soon as I save the newly created file and hit **Verify**, I always get:

```
/var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/APP.ino.elf
--nostdlib -L/Users/pk/Library/Arduino15/packages/sduino/hardware/stm8/0.3.3/STM8S_StdPeriph_Driver/lib -L/var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799 -L/Users/pk/Library/Arduino15/packages/sduino/tools/sdcc/build.10088/share/sdcc/lib/stm8 --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/sketch/APP.ino.cpp.o /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/../arduino_cache_648542/core/core_sduino_stm8_stm8sblue_7047d21421a119d005adccc6608c115c.a /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/core/main.c.rel -lSTM8S103 -lstm8 --out-fmt-elf -o /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/APP.ino.elf
cmd: /Users/pk/Library/Arduino15/packages/sduino/tools/sdcc/build.10088/bin/sdcc --nostdlib -L/Users/pk/Library/Arduino15/packages/sduino/hardware/stm8/0.3.3/STM8S_StdPeriph_Driver/lib -L/var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799 -L/Users/pk/Library/Arduino15/packages/sduino/tools/sdcc/build.10088/share/sdcc/lib/stm8 --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/sketch/APP.ino.cpp.rel /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/../arduino_cache_648542/core/core_sduino_stm8_stm8sblue_7047d21421a119d005adccc6608c115c.lib /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/core/main.c.rel -lSTM8S103 -lstm8 --out-fmt-elf -o /var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/APP.ino.elf
?ASlink-Error-<cannot open> : "/var/folders/bp/6gs20d4n1c33z8cmrvz92p2h0000gn/T/arduino_build_876799/core/main.c.rel"
exit status 1
Error compiling for board STM8S103F3 Breakout Board.
```

AndySymons, 2018-09-04:
Me too. Did you find a solution yet?

tenbaht, 2018-12-06:
It looks like this is not limited to OSX and I start to wonder why not more people are complaining about this.
Tested with 1.8.5 and 1.8.7 on OSX and Linux shows the same result:
- open a sketch, verify (compile): ok
- File->save as... and change a different name
- verify (compile): fails.

Workaround: Go to File->Preferences and uncheck the option "Aggressively cache compiled core"

The reason for this error message is that main.rel can't be used from to the regular core library but needs to be linked directly from the compiled object file. Looks like I have to find a way around this.

closed 2018-12-07 by tenbaht

tenbaht, 2018-12-07:
I just released v0.4.0 that fixes this issue.
The root cause was that there was not external reference to main, so the linker never bothered to pull main.rel in from the libcore. As a workaround I used to leave main.rel out of libcore and link it separately. That guaranteed that it is pulled in.
But when the current sketch is renamed and the aggressive caching is active, the Arduino build system moves the libcore (and only the libcore) into an archive/build cache directory and deletes the old build directory - that contained the main.rel. Resulting in a linker failure on the next compile.
New way to deal with this: I merged all of main.c into wiring.c. This file is always pulled in because it contains the ISR for TIM4 (the millis() timer). This way main() is always pulled in, even from libcore and without any special linker parameters.