tenbaht/sduino #53: AnalogReadSerial not compiling

TurtleForGaming, 2018-04-18:
The code:
```
/*
  AnalogReadSerial

  Reads an analog input on pin 0, prints the result to the Serial Monitor.
  Graphical representation is available using Serial Plotter (Tools > Serial Plotter menu).
  Attach the center pin of a potentiometer to pin A0, and the outside pins to +5V and ground.

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/AnalogReadSerial
*/

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(1);
  // print out the value you read:
  Serial.println(sensorValue);
  delay(1);        // delay in between reads for stability
}
```

does not compile a give me this error:

> Arduino : 1.8.5 (Windows 10), Carte : "STM8S103F3 Breakout Board"
> 
> bash.exe: warning: could not find /tmp, please create!
> C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\AnalogReadSerial.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
> Mark re12:C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\AnalogReadSerial.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
> cpp gefunden
> bash.exe: warning: could not find /tmp, please create!
> C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\AnalogReadSerial.ino.cpp sketch\AnalogReadSerial.ino.cpp.o re2 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
> Mark re2:C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\cores\sduino -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\variants\standard -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1/STM8S_StdPeriph_Driver/inc -IC:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\AnalogReadSerial.ino.cpp -o sketch\AnalogReadSerial.ino.cpp.o
> cpp gefunden
> C:\Users\Samuel\AppData\Local\Temp\arduino_modified_sketch_374854\AnalogReadSerial.ino:16: error 25: Structure/Union expected left of '.->'
> 
> C:\Users\Samuel\AppData\Local\Temp\arduino_modified_sketch_374854\AnalogReadSerial.ino:24: error 25: Structure/Union expected left of '.->'
> 
> exit status 1
> Erreur de compilation pour la carte STM8S103F3 Breakout Board
> Bibliothèque non valide trouvée dans C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library : C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library
> Bibliothèque non valide trouvée dans C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library : C:\Users\Samuel\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.1\libraries\MAX6675-library

zoomx, 2018-04-25:
You should use the examples that are for the STM8 boards.
The same example has an
#include <Serial.h>
and the print is written in this way
Serial_println_u(sensorValue);

mcuOne, 2018-09-21:
I have tried the code, but nothing comes out at the port PD4 and PD5. Which ports uses the Lib?

zoomx, 2018-09-22:
On a stm8s103f3 should be PD5 TX and PD6 RX

tenbaht, 2018-10-08:
pinout diagram: https://tenbaht.github.io/sduino/hardware/stm8blue/

closed 2018-10-08 by tenbaht