tenbaht/sduino #55: fatal error: Servo.h: No such file or directory

segatecm, 2018-05-27:
Windows 10, Arduino 1.6.7
Install from Boards Manager, version 0.33

test code:

#include <Arduino.h>
#include <Servo.h>

Servo myservo;          // just a simple unsigned char to hold the channel-ID

Please help!

zoomx, 2018-05-29:
I loaded the sweep example.

There is not an #include <Arduino.h> so I believe you don't need it.
You should use Servo(myservo); instead of Servo myservo;

Anyway I get an error too
`D:\IDE\arduino\portable\packages\sduino\hardware\stm8\0.3.3\libraries\Servo\src\stm8\Servo.c:24:22: fatal error: irqtable.h: No such file or directory
`

segatecm, 2018-05-30:
My problem is Arduino IDE can not find the Servo.h when Sduino used. Maybe the sduino path not configured right.

zoomx, 2018-05-31:
Maybe it is the IDE that is now at 1.8.5
You can check easily using the portable version
Go here
[https://www.arduino.cc/en/Main/Software](url)
download the "Windows ZIP file for non admin install"
unzip in a folder maybe in C: named Arduino, for example.
Go inside the folder where is the Arduino.exe executable
Create a folder called portable.
Launch the IDE (so all necessary files and folders will be created) then install the sduino core
Check if the example compile.
If it compiles you have to upgrade the IDE.

segatecm, 2018-06-03:
Hi @zoomx 
I install the IDE 1.8.5, and reinstall the Sduino.

When I compile the Servo code, it report the other problem:

Multiple libraries were found for "Servo.h"
 Used: C:\Users\chenmin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Servo
 Not used: C:\Users\chenmin\Documents\arduino-1.8.5\libraries\Servo
exit status 1
Error compiling for board STM8S103F3 Breakout Board.

This time much Servo.h files were found.

zoomx, 2018-06-04:
As I wrote before the Servo library was not updated, in the repository there is the one that has that error fixed.

I downloaded the Servo.c from here
[https://github.com/tenbaht/sduino/tree/master/sduino/hardware/sduino/stm8/libraries/Servo/src/stm8](url)
Just click on file, then on RAW
or use the direct link
[https://raw.githubusercontent.com/tenbaht/sduino/master/sduino/hardware/sduino/stm8/libraries/Servo/src/stm8/Servo.c](url)
then save page and maybe rename it Servo.c

put here
C:\Users\chenmin\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\libraries\Servo\src\stm8\
and overwrite the one that is inside.

segatecm, 2018-06-04:
@zoomx , Thanks for reply.
But the link your gave is invalid now. Github return 404 error.

closed 2018-06-05 by segatecm