tenbaht/sduino #58: How-To for other STM8 devices and boards?

gicking, 2018-09-30:
Hello Michael,

just installed Sduino from the Arduino-IDE as described - no issue and it worked immediately. Wow, I am impressed!!! 

Now I would like to add new boards to Sduino. My first attempt is a custom board based on STM8S207 with a LOT of peripherals. Upload is via USB and stm8gal. My (commandline) toolchain already works, but how can I get it into the Arduino-IDE...? As I have never added a new board to the Arduino-IDE, I have a LOT of questions:

- is there a How-To for adding boards to the Arduino IDE?
  - which files need to be adapted, which folders added?
  - how to select the upload method (stm8gal or stm8flash)?
- besides the supported STM8S there exist several other variants, e.g. STM8L -->
  - how to install other [(patched) SPLs](https://github.com/gicking/STM8-SPL_SDCC_patch) besides the already used STM8S-SPL?
  - how to select the correct one in the IDE (by board selection)...?

Sorry for my many questions! But that would be really great :-)  For any advice thanks a lot in advance!

Regards,
Georg (gicking)

gicking, 2018-10-03:
hi again,

maybe I was a bit too fast...  Seems like the install via boardmanager, as described [here](https://tenbaht.github.io/sduino/), installs an old version. The version shown in board manager is 0.3.3, which should be the latest version...?  However, board "s8uno" and support for stm8gal is completely missing, also in "boards.txt" and "platform.txt". See attached comparison of the github content and local installation.

[image]
[image]
[image]
[image]

 Any idea what I do wrong? Thanks a lot in advance and kudos for your great project!!!

Regards, Georg

PS: I found some issues with stm8gal and Arduino IDE, namely hexfile import and wait for return on error. When I have everything up and running, I will update it on Github

tenbaht, 2018-10-08:
Hi Georg,

I am back again and will have time for this project again. Next thing will be an updated release using sdcc 3.8.0, that will include the changes done after 0.3.3.

gicking, 2018-10-08:
hi Michael,

welcome back  :-)

When updating sduino please also consider the recent update of stm8gal and my (and the other) pull requests. Looking forward to the update and kudos to this great project!

Have a great day,
Georg