tenbaht/sduino #59: Tool sdcc is not available for your operating system.

tuxd3v, 2018-10-06:
Hello I am on arm64 arch..
I have installed gcc from sources, on:
/opt/sdcc
I made sdcc testes its ok.

But when I have your json config board: package_sduino_stm8_index.json
arduino IDE trows me a error:
"
 Tool sdcc is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool sdcc is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.base/java.lang.Thread.run(Thread.java:844)
Caused by: java.lang.Exception: Tool sdcc is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:85)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
 Tool sdcc is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool sdcc is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.base/java.lang.Thread.run(Thread.java:844)
Caused by: java.lang.Exception: Tool sdcc is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:85)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
 Tool sdcc is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool sdcc is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.base/java.lang.Thread.run(Thread.java:844)
Caused by: java.lang.Exception: Tool sdcc is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:85)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
"
I suspect its because of:
                    "toolsDependencies": [
                        {
                            "name": "STM8Tools",
                            "version": "0.3.3",
                            "packager": "sduino"
                        },
                        {
                            "name": "sdcc",
                            "version": "3.8.0",
                            "packager": "sduino"
                        }

how can I surpass this? so that board will be recognized?

thanks in Advance

tenbaht, 2018-10-08:
Supporting a processor architecture for the automated IDE install requires adding a link to a package containing the sdcc for that architecture to the package file. So far I didn't do that for ARM, but it sounds really interesting.  Once you have a package for sdcc and for the tools (avrdude etc.) it is not too hard. Is this for the Raspi 3?

I understand that you compiled sdcc yourself. There is a precompiled snapshot build for aarch64 on http://sdcc.sourceforge.net/snap.php . Would that work for you?

I am definitely interested in adding support for the Raspberry. Fall is the time of year where I might have the change to find time for sduino again.

tuxd3v, 2018-10-09:
Hello,
first of all, thanks for your response.

I am on RockPro64, on Ubuntu bionic.
I think it would be like raspberrypi, because I have 4 A53 cores,

It could be nice to have a version for arm64, :)
I haven't tried the pre-compiled version, I compiled one, but I made some tests, and it seems ok.
I installed it on /opt/sdcc
I am mostly interested in smt8, and I was trying to see if possible to program and flash, from Arduino, with your libraries, and stm8flash.

I revised my process, following your instructions here:
https://tenbaht.github.io/sduino/usage/manual-install/

I linked stm8flash, and sddcc to 
/home/rock64/Arduino/hardware/sduino/tools
also I created a aarch64 folder with a link also to smt8flash

And The board appears in Tools->Board{ stm8s105c6 Discovery Board ] selected..

Could be, ..that I solved the problem,
and are ready to blink a led or something?