tenbaht/sduino #61: bash.exe: warning: could not find /tmp, please create!

GeGeLv, 2018-10-07:
bash.exe: warning: could not find /tmp, please create!

wilhelmzeuschner, 2018-10-21:
I have the exact same problem. I just followed the instructions for the automatic install.

aryeguetta, 2018-11-15:
I have the same problem, win 8.1, ide 1.8.5, automatic install, Sduino 0.3.3. /tmp not exist. please assist.

1Sara1, 2018-12-02:
I figured out a fix for the /tmp directory error:

Create a tmp directory one directory before bash.exe:
C:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\STM8Tools\2017.11.13\tmp
**replace 'a' with your username**

**Now the /tmp error is gone there are more errors compiling blink example:**

C:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\Blink.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
Mark re12:C:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -c -Ddouble=float -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\Blink.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
C:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc sketch\Blink.ino.cpp sketch\Blink.ino.cpp.o re2 -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include
Mark re2:C:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/bin/sdcc -MMD -c -Ddouble=float -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\cores\sduino -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3\variants\standard -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/inc -IC:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/include sketch\Blink.ino.cpp -o sketch\Blink.ino.cpp.o
cpp gefunden
--nostdlib -LC:\Users\a\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.3.3/STM8S_StdPeriph_Driver/lib -LC:\Users\a\AppData\Local\Temp\arduino_build_910539 -LC:\Users\a\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.10088/lib/stm8 --code-size 8192 --iram-size 1024 -mstm8 -DSTM8S103 sketch\Blink.ino.cpp.o C:\Users\a\AppData\Local\Temp\arduino_build_910539/..\arduino_cache_907211\core\core_sduino_stm8_stm8sblue_027bf22e8dbe6c4787652511b4ec2d77.a C:\Users\a\AppData\Local\Temp\arduino_build_910539/core/main.c.rel -lSTM8S103 -lstm8 --out-fmt-elf -o C:\Users\a\AppData\Local\Temp\arduino_build_910539/Blink.ino.elf
?ASlink-Error-<cannot open> : "C:\Users\a\AppData\Local\Temp\arduino_build_910539/core/main.c.rel"

exit status 1
Error compiling for board STM8S103F3 Breakout Board

C:\Users\a\AppData\Local\Temp\arduino_build_910539
Sub directories Core and Libraries are empty.
none of the files the compiler is looking for.

tenbaht, 2018-12-06:
Hm. Strange.
I used a fresh virtual machine with Windows 7 from https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/ for VirtualBox and installed Arduino-1.8.7.

I can reproduce the bash message about missing /tmp and I can confirm that 1Sara1's workaround of creating a directory named tmp inside of STM8Tools fixes the message issue.

And then everything works for me. Compilation and upload, all fine.
@1Sara1: Does this compilation error always happen? I sounds more like a case of a lost temp directory to me. Something that fixes itself by closing and re-opening the Arduino-IDE. Or maybe a full harddrive that stops the systems from saving the compilation results.

commit 5d454b6f (2019-05-10): Merge branch 'busybox' into development (fix #61)