tenbaht/sduino #62: Any plans to add C++ Support?

bmitov, 2018-10-12:
I am looking if I can add support for the STM8 in Visuino. I already have added some support for STM32 and expanding it, and support most other modules - Nordic, Teensy, PIC32, AVR and many more.
To add the support I will need to have C++ however since all Visuino libraries are written in C++ using the advanced C++ features to reduce dramatically the generated code. The automatic code generation is specifically optimized to generate shockingly small memory footprint, but for this it requires C++.
Is there any chance that C++ support will be available in the future so I can add support in Visuino?
https://www.visuino.com

bmitov, 2018-10-12:
BTW: When I add it to Visuino, it will automatically get support all the Visuino libraries for all the hundreds of sensors, Displays and other modules that I have written ... well all written in highly optimized C++ (Much smaller footprint than the traditional Arduino libraries ;-) )

tenbaht, 2018-10-15:
Unfortunately not. C++ is a basic requirement for Arduino as well and the lack of a C++ compiler for the STM8 is the whole point of this project here - it is a C port of a C++ based development environment. This requires some compromises and it is not a drop-in replacement, but it is pretty close.

closed 2018-10-15 by tenbaht

bmitov, 2018-10-15:
What is the reason C++ is not supported?

ElDominio, 2018-12-04:
There is no compiler for C++ code for the STM8 line.

ElDominio, 2018-12-04:
IAR is not open source

zoomx, 2018-12-05:
STM has released an STM8 core for Arduino IDE some months ago. Since there is not any free compiler it uses Cosmic but you have to register and get a free license for one year. I believe that when the year license end you have to request again.