tenbaht/sduino #63: Can sduino use sleep mode?

segatecm, 2018-10-23:
I have a project used sduino. I need the STM8S in sleep mode and wake up by GPIO interrupted.
Can I use set_sleep_mode (SLEEP_MODE_PWR_DOWN) or something to do it? 
Thanks.

tenbaht, 2018-11-15:
Sorry, that is not implement yet. But it would be very useful, so it should go on the list.
Could you maybe post your findings if you manage to get the sleep mode working? I would love to include that into the system.

segatecm, 2018-11-15:
Hi tenbaht. I am a new hand to stm8.  I want to make some help but I can't. If you make it work I can try it, and thanks for your work.

closed 2018-11-15 by segatecm

jay-ramani, 2022-03-29:
@segatecm, did you find a workaround? Asking since the issue is closed. Not sure if it has been implemented.

jay-ramani, 2022-04-16:
Ok, I found something - use wfi (wake for interrupt) or halt.

Hope this helps others tripping on the same.

chirag-uAtech, 2023-12-20:
> Ok, I found something - use wfi (wake for interrupt) or halt.
> 
> Hope this helps others tripping on the same.

I am trying to implement the same. But am unable to find any documentation on this. It would be very helpful if you could share some easy steps for a newbie to implement this.