tenbaht/sduino #67: Any idea to add support for stm8L.

TiNredmc, 2018-11-20:
Hello.Im very new to the stm8 .Normally I'm mostly working on stm32 and avr_arduino .but this time I got some of stm8l151f3 mcu from chinese seller on eBay.And I have some suggession for add support for stm8l.Im also working on that too.by replace the stm8s SPL with the stm8l one .I also done some board.txt changes and patched the stm8l SPL .but It stuck on some error ,maybe about the syntaxes .So will look into them later.It would be great idea to add support for the stm8l .

tenbaht, 2018-11-20:
Yes, that would be interesting. Did you notice the build script to cut the SPL files into small pieces before compiling? That is needed as a workaround to help the linker to pull only the needed functions instead of the whole library. It shouldn't be hard to adopt that for the STM8L files. I could help if you need that.

tenbaht, 2018-11-20:
Oh, please re-open this issue if needed.

gicking, 2018-12-08:
sorry to re-open this issue! But what is the general procedure to add support for a STM8 variant other than STM8S or STM8A. SDCC patches for all available SPLs (including respective device headers) are [available](https://github.com/gicking/STM8-SPL_SDCC_patch), but what to do with them in sduino...?

If I can help (in my limited way) please let me know!

tenbaht, 2019-01-15:
After finishing the work on the SPL splitter I started to add some STM8L related files to a new branch [add-stm8l](https://github.com/tenbaht/sduino/tree/add-stm8l).

This is only the beginning of some basic scaffolding. It is incomplete and does not compile, but it should be a good start.

    - pins_arduino.h does not exist yet
    - no support for any hardware at all (not even GPIO)
    - no interrupt routines at all

I don't own any STM8L hardware, so any help is highly appreciated!

@TiNredmc: Maybe that can help you with your library problem? I included a precompiled library STM8L15X_LD for low-density-devices to the repro that should work with your STM8L151F3.

TiNredmc, 2019-01-17:
I use this board : https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F323104121862
I also modded the pins_arduino.h

tenbaht, 2019-01-17:
Interesting! That seems to be the only source for that little board. Since it is no a very common one I would opt for a generic pin mapping. Did you upload your pins_arduino.h somewhere? or could you attach it here?
I would be very interested in including your additions for the 8L series.

TiNredmc, 2019-01-22:
Here is it.
[pins_arduino.zip](https://github.com/tenbaht/sduino/files/2783276/pins_arduino.zip)

Jugulaire, 2019-04-08:
Can it work with this board ? https://www.ebay.com/itm/2PCS-STM8S003-DIP-20-Small-Package-development-tool-STLink/322985694957?hash=item4b337296ed:g:boAAAOSwusdaUqs7

tenbaht, 2019-04-09:
I didn't check that board in detail, but the S003 is almost identical to the S103. The only difference is the EEPROM size (128 bytes vs. 640 bytes). So yes, if it works with the S103 it will work with the S003.

 

(I am travelling right now and won't be back to STM8 things before end of April)

 

  Michael

 

 

Gesendet: Montag, 08. April 2019 um 18:38 Uhr

andrewsclapp, 2020-05-07:
Hello Michael,

I just wanted to say that I appreciate all the work you're doing here.  I have been working on a project with the STM8S103/003 chips, but we've recently decided to go after the STM8L051 as it has 12 bit ADC and more current capabilities that better suit our needs.  In that light, I was able to make some progress on getting a library file built for use with SDCC using the patch here ( https://github.com/gicking/STM8-SPL_SDCC_patch ) and then using your method from here ( https://tenbaht.github.io/sduino/developer/spl/ ).  I still have testing to do, but I wanted to say thank you for all your work.

-ASC