tenbaht/sduino #69: IDE expects file "board.txt" at different location

gicking, 2018-11-25:
hello Michael,

I have just cleanly forked sduino in Github and re-installed it locally following your [instructions](https://tenbaht.github.io/sduino/usage/manual-install/) - much cleaner now :-)   But when I open the library manager in the Arduino-IDE (1.8.6) I get the following warning (in German): 

`Es konnte keine boards.txt in /home/georg/Dokumente/Arduino/hardware/sduino/tools gefunden werden. Ist sie noch von einer Version vor Arduino 1.5?`

Is this important or can I ignore this? Compiling works, uploading not yet (see separate issue)

Thanks in advance!

Georg

per1234, 2018-11-25:
The Arduino IDE expects this hardware package folder structure:
```
{sketchbook folder}
|_hardware
   |_{vendor name}
      |_{architecture name 1}
      |   |_boards.txt
      |   |_etc...
      |_{architecture name 2}
          |_boards.txt
          |_etc...
```
This allows one vendor to have multiple architectures (e.g. arduino:avr, arduino:samd, arduino:sam). So if you have this folder structure:
```
/home/georg/Dokumente/Arduino
|_hardware
   |_sduino
      |_stm8
      |_tools
```
the Arduino IDE expects the `tools` folder to contain the hardware package for the sduino:tools architecture. But `tools` does not contain a hardware package, so the IDE displays a warning. The `stm8` folder does contain the valid hardware package.

gicking, 2018-11-25:
does an empty "boards.txt" avoid this warning? That would avoid stupid questions like this... ;-)

per1234, 2018-11-25:
It does. The only side effect I can see is that this causes an extra horizontal divider line to be added to the **Tools > Board** menu as a separator for the empty hardware package. I don't think that's a big deal and most people probably wouldn't even notice it.

If this is done, I recommend adding a comment to the dummy boards.txt file to explain its purpose.

tenbaht, 2018-11-26:
Good idea. ESP32 installs the tools in hardware/espressif/esp32/tools - the equivalent would be hardware/sduino/stm8/tools
Maybe I should try that. But getting 0.4.0 out is first.

tenbaht, 2018-11-30:
Yes, I would like to change the file structure. But that might break things, so I postphone that for the time after 0.4.0. I am getting closer...