tenbaht/sduino #72: digitalWrite doesn't work after analogWrite

Gei0r, 2018-12-26:
I have an issue with `analogWrite()` on the stm8blue board.
A call to `analogWrite()` with 255 should set the pin constantly to `HIGH` (disabling PWM). But that only works sometimes if there was another analog (PWM) value written there before:

    analogWrite(2, 1);
    analogWrite(2, 255);  // pin permanently stays off almost always

    analogWrite(2, 64);
    analogWrite(2, 255);  // works about 25 % of the time, pin stays off otherwise
    
    analogWrite(2, 128);
    analogWrite(2, 255);  // works about 50 % of the time, pin stays off otherwise

    analogWrite(2, 192);
    analogWrite(2, 255);  // works about 75 % of the time, pin stays off otherwise

    analogWrite(2, 254);
    analogWrite(2, 255);  // works almost 100 % of the time

Exchanging the second `analogWrite()` with `digitalWrite()` doesn't make a difference.

It seems that the pin gets "stuck at `LOW`" if the PWM is disabled while the pin is driven low by PWM.

Maybe `turnOffPWM()` in `wiring_digital.c` is insufficient?

    static void turnOffPWM(uint8_t timer)
    {
    	*((unsigned char *) ccmrx[timer-1]) &= ~TIM1_CCMR_OCM;
    }

tenbaht, 2018-12-27:
Thank you for the report. I will have a look at this after new year's.

Gei0r, 2018-12-30:
I took a look into the datasheet myself. As I suspected, the code in `turnOffPwm()` only "freezes" the timer comparator, but it does not disable the PWM output driver.

For that, `CCiE` / `CCiNE` in `CCER` need to be cleared. It's fixed in the referenced pull request.

[image]

PR #73 (2019-01-08): Correctly disable PWM output in digitalWrite()
Fixes #72.

The old line in turnOffPWM() which freezes the timer compare channel is not strictly necessary. I left it in because it seems cleaner, but of course there's a small performance cost.
commit 91a7d7d5 (2019-01-08): Correctly disable PWM output in digitalWrite()

Fixes #72.
closed 2019-01-08 by tenbaht