tenbaht/sduino #74: No bale to upload sketch to STM8S105K4T6 Breakout board

aryeguetta, 2018-12-31:
Hi, 
I have just bought two STM8S105K4T6 Broads and I have connected them to my PC using STLink-V2 with the latest Firmware installed on it.

I already installed the sduino boards version 0.4.0. meaning I am up-to-date with both (so I think I am)
Trying to upload a simple "Blinky" Sketch to the board with NO luck at all :(
**Boards is:** STM8S105K4T6 Breakout Board
**Upload Method:** STLink\V2
So I think I am OK with the settings, correct? Not much to choose from
Connecting the board with 3V3 and GND to the STLink-V2 board. and RST to RST and SW from the board to the SWDIO (also tried SWIM with no luck). This part I am not sure if I am doing correct - But I am trying almost everything.. No luck!

I press the button on the board to set it to programming mode (So I think that is what I need to do - since it also DOES NOT work when I don't press the Button) eather way - It does NOT work at all - with me pressing the button or not.

I am getting lots of errors trying to upload the simple blink sketch and Nothing is working.
At the end I get:

> Determine FLASH area
> libusb: warning [init_device] could not get node connection information for device '\\.\USB#VID_046D&PID_C52B&MI_02#8&3A8DFF1F&0&0002': [87] The parameter is incorrect.
libusb: info [cache_config_descriptors] could not access configuration descriptor (dummy) for '\\.\USB#VID_1A40&PID_0101#7&3A515E2&0&2': [31] A device attached to the system is not functioning.
Due to its file extension (or lack thereof), "C:\Users\JuliaG\AppData\Local\Temp\arduino_build_139416/Blink.ino.hex" is considered as INTEL HEX format!
2767 bytes at 0x8000... Tries exceeded
An error occurred while uploading the sketch 

I have TWO STLink-V2 and Two STM8S105K4T6 borads - and Nothing works on both!
I am sure that I am doing something wrong and I hope that you can assist me.

My main concerns are:

- WHERE should I connect the SW from the STM8S105K4T6 on the STLink?
- Should I press the BUTTON on the board or Shouldn't I?
- When I press it - Should it be pressed all the time when uploading or only once and hit upload?

Please assist. I want to make these boards work on my Arduino IDE :)
Thank you!
Aryeduino

tenbaht, 2019-01-08:
I looks like you are using Windows. Did you install the ST-Link-Driver from the ST Website? The driver needs to be installed before you connect the ST-Link. See [https://tenbaht.github.io/sduino/hardware/flashtool/](https://tenbaht.github.io/sduino/hardware/flashtool/)

You will need the SWIM connection, SWD is STM32 only. Using ST-Link you don't need to press any buttons, programming mode is activated automatically. Manual reset is only important for the serial bootloader connection using stm8gal.

wilhelmzeuschner, 2019-02-19:
I am facing the same issue, I have the driver installed and have updated the ST-Link V2 Firmware.
The connections are made according to your specifications / the markings on the board.
Board: STM8S103F3
IDE Version: 1.8.8
Which uploader do I have to select in the "Tools" menu in the Arduino IDE is this setting even relevant since I don't have to select a COM port?

Error message:
```
Sketch uses 2505 bytes (30%) of program storage space. Maximum is 8192 bytes.
Global variables use 70 bytes (6%) of dynamic memory, leaving 954 bytes for local variables. Maximum is 1024 bytes.
Determine FLASH area

libusb: error [init_device] device '\\.\USB#VID_1532&PID_011B&MI_01#7&312D0923&0&0001' is no longer connected!
libusb: error [init_device] device '\\.\USB#VID_1532&PID_011B&MI_02#7&312D0923&0&0002' is no longer connected!
Due to its file extension (or lack thereof), "C:\Users\user\AppData\Local\Temp\arduino_build_126147/Blink.ino.hex" is considered as INTEL HEX format!
2605 bytes at 0x8000... Tries exceeded
the selected serial port 2605 bytes at 0x8000... Tries exceeded
 does not exist or your board is not connected
```

tenbaht, 2019-03-02:
Hi Wilhelm,

Tools->upload method should be stlink/v2 if you don't want to use the serial bootloader. For the older versions of the IDE it was important to select a COM port beforehand, even if it is not used. IMHO that got fixed with 1.8.8 but it might be worth a try.

The error message sounds more like an USB issue. How about the current consumption? Is there anything connected to the breakout board? If too much current is consumed the USB controller automatically shuts down the port. The limit might be as low as 100mA, depending on the configuration of your programming device. The current includes the ST-Link device, the breakout board and everything connected from there.

Maybe try to power the breakout board from a separate power supply and don't connect the Vcc line from the ST-Link.

wilhelmzeuschner, 2019-03-02:
Michael (@tenbaht ) thanks for your reply!
I was able to upload to my board.
Truth be told, I probably didn't have very good connections between the jumper wires and my board.
Just as a potential reference to others / myself: Don't use jumper wires that don't have significant friction when plugged in.
As you stated, _it is not necessary_ to actually select a serial port.

Curiously the programming output still includes the error message(s) I was getting before.
If I select a COM port when uploading (which is not neccesary) I get only one `libusb: error [init_device] device` error.
No COM port selected:
```
Sketch uses 2505 bytes (30%) of program storage space. Maximum is 8192 bytes.
Global variables use 70 bytes (6%) of dynamic memory, leaving 954 bytes for local variables. Maximum is 1024 bytes.
Determine FLASH area

libusb: error [init_device] device '\\.\USB#VID_1532&PID_011B&MI_01#7&312D0923&0&0001' is no longer connected!
libusb: error [init_device] device '\\.\USB#VID_1532&PID_011B&MI_02#7&312D0923&0&0002' is no longer connected!
Due to its file extension (or lack thereof), "C:\Users\manua\AppData\Local\Temp\arduino_build_185623/Blink.ino.hex" is considered as INTEL HEX format!
2605 bytes at 0x8000... OK
Bytes written: 2605
```

tenbaht, 2019-03-02:
I am glad to hear. So I can close this issue? The problem was just bad wiring?

BTW: The USB Vid/Pid refers to a mouse, not the ST-Link adapter: http://www.devicekb.com/usb-vendor/1532/device/011B.php
Does that make any sense? Do you own such a mouse? Maybe there is a vendor specific (bloatware-) driver that causes the libusb error message?

wilhelmzeuschner, 2019-03-03:
In my case the issue is resolved.
USB#VID_1532&PID_011B&MI_02#7&312D0923&0&0002' does indeed refer to another device, my keyboard.

tenbaht, 2019-03-03:
Ok, great. So I declare this a wiring problem and close the issue!

closed 2019-03-03 by tenbaht

Murugesh-Hobbyist, 2024-11-09:
Hi I too having same problem and cross checked the wiring all perfect. driver is installed too. tried both V2 and V2.1 in programmer. Please advice.

below code:

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000); 
  digitalWrite(LED_BUILTIN, LOW); 
  delay(1000);    
}

Serial Monitor output is below:
Sketch uses 1703 bytes (20%) of program storage space. Maximum is 8192 bytes.
Global variables use 72 bytes (7%) of dynamic memory, leaving 952 bytes for local variables. Maximum is 1024 bytes.
Determine FLASH area

Due to its file extension (or lack thereof), "C:\Users\murug\AppData\Local\Temp\arduino_build_715628/Blink.ino.hex" is considered as INTEL HEX format!
1821 bytes at 0x8000... Tries exceeded
An error occurred while uploading the sketch