tenbaht/sduino #75: Ds18b20 sensor with Sduino : Example removed

Jugulaire, 2019-03-10:
Hi, 

I'm working on Ds18B20 based thermometer.
In older version on Sduino we have a nice Example of DS18B20 usage but i cannot find it anymore. 
Moreover, the documentation about I2C and Wire is not clear at all. 
Even with example i cannot figure out where to plug my DATA pin..
You mention a bit the pin 3/4 in the breakout board documentation but nothing more.. 

Can you help me to clarify where to plug my DATA pin ? or how to define it ? 
Do you have retired the Example with DS18B20 ?

Jugulaire, 2019-03-11:
Nevermind, i've figured ou why. I made a confusion between Wire and oneWire protocol which is totally différent.

Jugulaire, 2019-03-11:
I'm starting to port my own version of OneWire protocol to Sduino. 
I'm working on the integration for selecting the pin and writing a special lib for DS18B20. 
Everything work fine with onewire on this platform.

abdul-rehman-2050, 2022-10-01:
did you wrote that library? because I am struggling the same problem and couldn't find any solution over internet. I tried bunch of different code written for other boards but no one worked for me.