tenbaht/sduino #76: sdcc compiler sseg error with 4.0

jsadv, 2019-04-12:
[code]
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
[/code]

ASlink-Warning-No definition of area SSEG
exit status 1
Error compiling for board STM8S105K4T6 Breakout Board.

loads default blink tho....???????

closed 2019-04-12 by jsadv

JonasCz, 2020-09-23:
I have the same problem: it compiles bigger code fine, but empty program or LED blink fails with;

`ASlink-Warning-No definition of area SSEG`

Did you ever fix this, and how?

tenbaht, 2020-10-14:
This happens when there is no *.pde or *.ino file in the project directory and only *.c files are compiled.

This message means that main.c is not pulled in by the linker because there was no reference to main() anywhere. When processing *.pde/ino files `wrapper/sdcc.sh` (for IDE builds) or `sduino.mk` (for Makefile builds) adds a reference to main:
```
/* add a dummy reference to main() to make sure to pull in main.c from the core library */
void main(void);
void (*dummy_variable) () = main;
```
If there is no pde/ino file the user has to make sure main.c is pulled in by the linker or define its own main().

Possible ways to pull in main.c:
a) Use Serial_begin(): This references the variable runSerialEvent which in turn pulls in main.c (some overhead)
b) reference the variable runSerialEvent yourself: `begin(){runSerialEvent=0;}` (less overhead)
c) add a reference to main() like above. (minimal overhead)
d) define your own main() function.

This issue keeps popping up over and over again. This might require a FAQ entry.

commit 3d685a8c (2020-10-15): add FAQ, minor spelling fixes, fix #76