tenbaht/sduino #78: Request: board support for stm8s003f3p6

aghfars, 2019-09-02:
please add support for stm8s003f3p6 on arduino ide

mcer12, 2019-09-04:
stm8s003f3p6 is already supported, have you tried it?

aghfars, 2019-09-04:
[image]
no it isn't here
where can i found it?

mcer12, 2019-09-04:
use stm8s103

aghfars, 2019-09-04:
i need to program stm8s003 not stm8s103
you mean i use stm8s103 to prgram stm8s003 ?

mcer12, 2019-09-04:
Yes

aghfars, 2019-09-09:
yes its worked.
but my stm8s103f3p6 give me this error 
Determine FLASH area

libusb: error [init_device] device '\\.\USB#VID_09DA&PID_5208&MI_02#6&28F4B66C&0&0002' is no longer connected!
Due to its file extension (or lack thereof), "C:\Users\AHMADR~1\AppData\Local\Temp\arduino_build_324501/Blink.ino.hex" is considered as INTEL HEX format!
1821 bytes at 0x8000... Tries exceeded
the selected serial port 1821 bytes at 0x8000... Tries exceeded
 does not exist or your board is not connected

aghfars, 2019-09-09:
and i try with stvp and give me that error 
> Reading  PROGRAM MEMORY area...
Error : Cannot communicate with the device !
Check the SWIM cable connection and check all the needed pin connections on the SWIM connector.

If the application code uses Swim Disable and Reset pin as Output or has disabled SWIM Clock Divider:
Try Now to SWITCH OFF and ON the application Power Supply while NRST Reset pin is forced low.

Error : < PROGRAM MEMORY reading failed.

aghfars, 2019-09-09:
what is the problem?
i double check the wiring. its OK.
how can i fix this?

tenbaht, 2020-02-06:
The '003 and the '103 are almost identical. The only difference I am aware of is the size of the EEPROM (128 bytes vs. 640 bytes). Just set the board manager for STM8S103 and start compiling. (and don't try to use EEPROM locations > 127 or  > 0x7f).
Since this question keeps popping up I added this to the project website:
https://tenbaht.github.io/sduino/hardware/stm8s003/

closed 2020-02-06 by tenbaht