tenbaht/sduino #79: I2C slave mode

mcer12, 2019-09-04:
Any plans to implement I2C slave mode? For such low tier device it would be very nice!

tenbaht, 2020-02-06:
I2C slave mode, I2C interrupt support and SPI interrupt support are on my todo list.
Unfortunately, I am quite short on time for this project, so I have to focus on the things I am actively using. So don't hold your breath :-(

liwenzhuang, 2020-07-09:
When can I use the IIC machine?Thank you very much

stefaandesmet2003, 2021-02-01:
I committed a i2c slave implementation here : https://github.com/stefaandesmet2003/sduino
Feel free to use & test

mcer12, 2021-02-03:
@stefaandesmet2003 looks nice, maybe you can make a PR from it? This is a very needed function for many!

engr-mujahidabdullah, 2021-04-11:
@stefaandesmet2003  i replace your repo with default and still Wire_beginSlave(int) not compiled give error. Kindly guide me in this regards.

mcer12, 2023-02-18:
@stefaandesmet2003 Two years later I revisit this issue :) I have bunch of STM8 at home from the time when they were very cheap but the best use for them - I2C slave - is not implemented... would you be able to merge your work with current version of the project in a new fork?

stefaandesmet2003, 2023-02-18:
I consider open source as a means of sharing my work, not of delegating my work to someone else

mcer12, 2023-02-18:
@stefaandesmet2003 You didn't share a source so you're the only one who can share it. Could've just said no instead of being petty.