tenbaht/sduino #80: Can you add HC-12 radiomodule support

diygoodies, 2019-09-05:
Hi! Thanks a lot for your work! Can you add HC-12 radiomodule support. It based on STM8S003F3P6?
[image]

diygoodies, 2019-09-05:
https://github.com/TG9541/stm8ef/wiki/Board-HC12

tenbaht, 2020-02-06:
Are you planning to write a new firmware for that board? Yes, that would certainly be possible with SDuino. To my understanding there is nothing special about this board on the STM8 side. The STM8S003 is almost identical with the STM8S103. Just choose 'stm8blue' and start developing. Just keep in mind that the '003 implements only 128 bytes of EEPROM compared to 640 byte for the '103.

The SPI library is still lacking interrupt support. Not sure if that would be a problem for the Si4463. Interrupt support is still on my list, but as you have noticed I am very short on spare time for this project, so don't hold your breath. But apart from that - go for it!

closed 2020-02-06 by tenbaht