tenbaht/sduino #81: the selected serial port 1821 bytes at 0x8000... Tries exceeded  does not exist or your board is not connected

TehseenHasan, 2019-09-11:
I'm trying to program **STM8S003f3 development board** using Arduino IDE v1.8.9. I am using **ST-Link V2 Programmer**. I have selected Board and Programmer correctly.

[image]

**I opened Blink Example and compile it. It compiled successfully.
But every time I try to Upload, the IDE gave error message and never uploads the sketch in STM8S003f3 board.**

**It give the following error message:**
[image]

Please help me to solve this issue. Thanks!

TehseenHasan, 2019-09-11:
### I found the solution...

First of all I have to remove Flash Protection of the STM8S003f3 board.
I used **ST Visual Programmer** tool to remove the protection.
(ST Visual Programmer Quick Tutorial: http://digital-wizard.net/stm8_tutorials/programming_stm8_using_swim)

**How to Remove the Flash Protection:**
1. Connect the board with ST-LINK programmer and plug the programmer in USB.
2. Open "ST Visual Programmer" application.
3. Select relevant configuration setting according to your board and programmer.
4. Go to **OPTION BYTE** tab.
5. Set ROP to Off and then press CTRL+P

Now you can Open Arduino IDE and upload the code easily without that error.

closed 2019-09-11 by TehseenHasan

tenbaht, 2020-02-06:
Just for future reference in case others land here via google: There is an easier solution: Just click on Tools->Burn Bootloader.
See #85 and https://tenbaht.github.io/sduino/hardware/stm8blue/#unlocking-a-write-protected-mcu