tenbaht/sduino #82: [Faltal] Not being able to compile the default sketch.

Kashouryo, 2019-09-15:
I can not compile the default empty sketch (that only contains setup() and loop()) with sduino.
I am using Arduino 1.9.0 on a windows 10 1803.

Here's the error log that I think is important to address this issue.

"C:\\Software\\Arduino IDE\\portable\\packages\\sduino\\tools\\STM8Tools\\2019.02.05/win/busybox" ash "C:\\Software\\Arduino IDE\\portable\\packages\\sduino\\tools\\STM8Tools\\2019.02.05/wrapper/sdcc.sh" "C:\\Software\\Arduino IDE\\portable\\packages\\sduino\\tools\\sdcc\\build.11242/bin/sdcc" "C:\\Users\\think\\AppData\\Local\\Temp\\arduino_build_786390\\sketch\\sketch_sep15a.ino.cpp" "C:\\Users\\think\\AppData\\Local\\Temp\\arduino_build_786390\\sketch\\sketch_sep15a.ino.cpp.o" re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 "-IC:\\Software\\Arduino IDE\\portable\\packages\\sduino\\hardware\\stm8\\0.5.0\\cores\\sduino" "-IC:\\Software\\Arduino IDE\\portable\\packages\\sduino\\hardware\\stm8\\0.5.0\\variants\\standard" "-IC:\\Software\\Arduino IDE\\portable\\packages\\sduino\\hardware\\stm8\\0.5.0/STM8S_StdPeriph_Driver/inc" "-IC:\\Software\\Arduino IDE\\portable\\packages\\sduino\\tools\\sdcc\\build.11242/include"
C:\Software\Arduino IDE\portable\packages\sduino\tools\sdcc\build.11242/bin/sdcc C:\Users\think\AppData\Local\Temp\arduino_build_786390\sketch\sketch_sep15a.ino.cpp C:\Users\think\AppData\Local\Temp\arduino_build_786390\sketch\sketch_sep15a.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Software\Arduino IDE\portable\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Software\Arduino IDE\portable\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Software\Arduino IDE\portable\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Software\Arduino IDE\portable\packages\sduino\tools\sdcc\build.11242/include
Mark re2:C:\Software\Arduino IDE\portable\packages\sduino\tools\sdcc\build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Software\Arduino IDE\portable\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Software\Arduino IDE\portable\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Software\Arduino IDE\portable\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Software\Arduino IDE\portable\packages\sduino\tools\sdcc\build.11242/include C:\Users\think\AppData\Local\Temp\arduino_build_786390\sketch\sketch_sep15a.ino.cpp -o C:\Users\think\AppData\Local\Temp\arduino_build_786390\sketch\sketch_sep15a.ino.cpp.o
cpp gefunden
C:\Users\think\AppData\Local\Temp\arduino_build_786390\sketch\sketch_sep15a.ino.c:2: syntax error: token -> '.' ; column 14

exit status 1

MateuszStudiosCompany, 2019-10-27:
It's looks like, they changed something in Arduino IDE. **For now, you have to download older version of IDE.**

Tested on Linux (Debian):
Same error: 1.9.0 (beta) x64
Working: 1.6.13 x64

The second is, for now randomly chosen. I'll try to check last working version. **Maybe it's just a problem od beta**.

Kashouryo, 2019-10-29:
Thank you~ I switched back to Arduino IDE 1.8.10
And It works~

closed 2019-10-29 by Kashouryo