tenbaht/sduino #83: Multiple libraries were found

polarbear82, 2019-10-30:
I get "Multiple libraries were found ..." for an rf transmitter library  compile but I have only one copy of that lib ( using Linux 18.04).
/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc sketch/HelloWorldTransmitter.ino.cpp preproc/ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/user/Arduino/libraries/RFTransmitter -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include
Mark re12:/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/user/Arduino/libraries/RFTransmitter -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include sketch/HelloWorldTransmitter.ino.cpp -o preproc/ctags_target_for_gcc_minus_e.cpp
cpp gefunden
/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc sketch/HelloWorldTransmitter.ino.cpp sketch/HelloWorldTransmitter.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/user/Arduino/libraries/RFTransmitter -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include
Mark re2:/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/cores/sduino -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/variants/standard -I/home/user/Arduino/libraries/RFTransmitter -I/home/user/.arduino15/packages/sduino/hardware/stm8/0.5.0/STM8S_StdPeriph_Driver/inc -I/home/user/.arduino15/packages/sduino/tools/sdcc/build.11242/share/sdcc/include sketch/HelloWorldTransmitter.ino.cpp -o sketch/HelloWorldTransmitter.ino.cpp.o
cpp gefunden
/home/user/Arduino/libraries/RFTransmitter/RFTransmitter.h:29: syntax error: token -> 'RFTransmitter' ; column 19
Multiple libraries were found for "RFTransmitter.h"
 Used: /home/user/Arduino/libraries/RFTransmitter
exit status 1
Error compiling for board STM8S103F3 Breakout Board.

polarbear82, 2019-10-30:
update , I get the same error with all Arduino libraries also with native one like Wire and etc.

tenbaht, 2020-02-06:
I guess you are using the RFTransmitter library from https://github.com/zeitgeist87/RFTransmitter ?
Like all regular Arduino libraries it is implemented as a C++ class, but SDduino is based on plain C and the original library files can't be used as-is.
Only the SDuino specific version will work, sorry.

closed 2020-02-06 by tenbaht