tenbaht/sduino #85: Write Protection Simple Solution

marcostech, 2019-11-12:
**Hello**,
I have experienced some issues while unlocking the stm8s103f3p6 blue breakout board on Windows, the solution is simple and i think you may want to reference it on the guide for unlocking the MCU, in the hardware section (https://tenbaht.github.io/sduino/hardware/stm8blue/).
**On arduino IDE you cant tell if your MCU is locked or it's a simple connection problem**, so its good to at least reference a "path" to solve the issue on Windows based systems where the users dont know how to use/or dont have bash cmds like Linux (**me**).
You already know how to unlock it but here it goes for the ones who dont:
If the board has write protection ON (Read Out Protection or ROP) you will need to install the **STVP (ST Visual Programmer**) to flash a new configuration that overwrites the Protection bytes and sets them off. 
To do so just do a simple registration on the ST website and donwload **STVP**. Install and open it > Configure ST Visual Programmer with Hardware(**ST-LINK**), Programimg Mode(**SWIM**) and Device(**STM8S103F3**)> select the "Option Byte" tab. Now the first item there is "**ROP**", put it OFF > Save the file as .hex > Import the **same file** > Program the MCU > and done. No more Arduino IDE errors.
While using the STVP with a locked board you will receive a **message saying its locked**, so to test it just try to read it after the flash has ended.

_**And pay attention to Chinese ST-link V2, they come with bad or no contacts on some pins, test them all out.**_

And thanks for your great job on SDUINO.

Best Regards,
Marcos Vinicius

tenbaht, 2020-02-06:
To unlock it using the Arduino IDE click on Tools->Burn Bootloader after selecting an STM8 based board and choosing the correct programmer type (ST-Link V2).

The name of this menu entry is not self-explanatory, but I couldn't find any way to change it or to add another entry with a better name. I updated the description on the stm8blue page to point this out.

closed 2020-02-06 by tenbaht

Lazlolozla, 2022-07-01:
Works!  Burn Bootloader of Arduino 1.8.19, kubuntu 20.04 LTS . STLINKv2 and STM8S103F3P6 made in China.
tanks!!!

commit a19e18d9 (2020-02-06): Mention how to unlock a STM8 using the Arduino IDE (fix #85)