tenbaht/sduino #87: can‘t reach package_sduino_stm8_index.json

fuwen0202, 2019-12-27:
Start the Arduino-IDE. In File->Preferences, Settings tab, enter
https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json

then，Arduino IDE show， error when download https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json 

try to open the url in chrome，failed too

fuwen0202, 2019-12-27:
所以我更换了github的文件抓取方式，将url改为以下地址后，即可
https://raw.githubusercontent.com/tenbaht/sduino/master/package_sduino_stm8_index.json

tenbaht, 2020-02-06:
Interesting find, thank you for mentioning! There should be 302 re-direct from the github.com URL to the raw.githubusercontent.com domain, but sometimes it is not working (for unknown reason). I updated the README accordingly.

commit a35d9144 (2020-02-06): change board manager URL from github.com to raw.githubusercontent.com domain (fix #87)
closed 2020-02-06 by tenbaht