tenbaht/sduino #91: PS/2 Library for STM8 board

tinkerBOY-git, 2020-02-24:
First I want to thank the author of this library for STM8 boards! It works great! My only issue is that it won't compile with available PS/2 libraries.

> PS2Keyboard.h:195: syntax error: token -> 'PS2Keyboard' ; column 17

tenbaht, 2020-02-24:
SDuino is based on plain C, not on C++ as the original Arduino. Since most Arduino libraries are written as a C++ class they need to be modified for use with SDuino first. Often this is not very hard, but this depends on the specific implementation.

closed 2020-02-24 by tenbaht

tinkerBOY-git, 2020-02-24:
> SDuino is based on plain C, not on C++ as the original Arduino. Since most Arduino libraries are written as a C++ class they need to be modified for use with SDuino first. Often this is not very hard, but this depends on the specific implementation.

Thank you!