tenbaht/sduino #92: Is attachinterrupt supported?

tinkerBOY-git, 2020-02-24:
Just new to sduino and would like to know if **attachinterrupt** is supported?

Pog3k, 2020-04-21:
Seems not. If you need interrupts use the STM8 lib functions which can be directly called.

For example PortC Pin4 
Init Pin:
 GPIO_Init(GPIOC, GPIO_PIN_4, GPIO_MODE_IN_FL_IT);
 EXTI_SetExtIntSensitivity( EXTI_PORT_GPIOC, EXTI_SENSITIVITY_RISE_FALL);

INTERRUPT_HANDLER(EXTI_PORTC_IRQHandler, 5) {
// your code here
}

in Winterrupts.c <- somwhere in the arduino path
Comment this line out.
//IMPLEMENT_ISR(EXTI_PORTC_IRQHandler,		INT_PORTC) /* EXTI PORTC */

Works for me.

leifclaesson, 2020-10-15:
Thanks for this, @Pog3k ! Your post helped point me in the right direction.

For future visitors:

I was seeing very erratic behavior and huge amounts of spurious interrupts, no matter what EXTI_SENSITIVITY mode i used.
I finally googled the [Official Documentation](https://documentation.help/STM8S-A-Standard-Peripherals-Drivers/group__EXTI__Exported__Functions.html#ga9116693abfcf1b224191e73dcabb2e45) and found that you must
**disableInterrupts()**
before calling EXTI_SetExtIntSensitivity
and
**enableInterrupts()**
again afterwards.
It seemed silly but I tried it and it totally solved the issue, my EXTI_PORT_GPIOA interrupt now works perfectly!

Also, you *don't* need to modify Winterrupts.c. You can use attachInterrupt(INT_PORTA & 0xFF,ISR,0); together with GPIO_Init and EXTI_SetExtIntSensitivity.

Here's my code for an interrupt on D2, meaning PA3, Port A Bit 3:
void ISR()
{
//your code here
}

void setup()
{
 pinMode(2,INPUT);

  GPIO_Init(GPIOA, GPIO_PIN_3, GPIO_MODE_IN_FL_IT);
  disableInterrupts();
  EXTI_SetExtIntSensitivity( EXTI_PORT_GPIOA, EXTI_SENSITIVITY_RISE_ONLY);  
  enableInterrupts();

  attachInterrupt(INT_PORTA & 0xFF,ISR,0);
}

closed 2020-10-30 by tenbaht

tenbaht, 2020-10-31:
Thank you for your experiments! attachInterrupt() is now fully supported: https://tenbaht.github.io/sduino/usage/interrupts/
It works like this:
```
void on_button_pressed(void)
{
	flag = 1;
}

void setup()
{
	pinMode(BUTTON, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(BUTTON), on_button_pressed, FALLING);
}
```

smartswitchsio, 2020-11-08:
I am getting bellow error while compiling example code at this link https://tenbaht.github.io/sduino/usage/interrupts/  for STM8S103F3, kindly check.
 
C:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\sketch_nov08c.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include -DARDUINO_LIB_DISCOVERY_PHASE
Mark re12:C:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include -DARDUINO_LIB_DISCOVERY_PHASE sketch\sketch_nov08c.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
C:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\sketch_nov08c.ino.cpp sketch\sketch_nov08c.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re2:C:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\Education\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\sketch_nov08c.ino.cpp -o sketch\sketch_nov08c.ino.cpp.o
cpp gefunden
C:\Users\EDUCAT~1\AppData\Local\Temp\arduino_modified_sketch_803111\sketch_nov08c.ino:21: warning 112: function 'digitalPinToInterrupt' implicit declaration
C:\Users\EDUCAT~1\AppData\Local\Temp\arduino_modified_sketch_803111\sketch_nov08c.ino:21: error 101: too many parameters 
exit status 1
Error compiling for board STM8S103F3 Breakout Board.

hmeijdam, 2020-12-03:
I am getting also compilation errors, so I checked the files that were changed with this issue and observed it is Arduino.h and Winterrupts.c I compared content of Winterrupts.c with my 0.5.0 installation and I think they are not part of the current sduino 0.5.0 released version, so I installed them manually in my Arduino IDE (1.8.13)

Then I still got compilation errors on the line
`attachInterrupt(digitalPinToInterrupt(BUTTON), on_button_pressed, FALLING);`

I looked in the Winterrupts.c source and saw that it is working with digitalPinToPort(pin)

so when I changed to: 
    `attachInterrupt(digitalPinToPort(BUTTON), on_button_pressed, FALLING);`

It works!
when connecting a pushbutton between A2 and GND the led blinks for 0.3 seconds when pressing the button

rtek1000, 2022-07-20:
> Thanks for this, @Pog3k ! Your post helped point me in the right direction.
>
> For future visitors:
>
> I was seeing very erratic behavior and huge amounts of spurious interrupts, no matter what [...]

Hello,

Does anyone know where this INT_PORTA reference is declared?

How can I declare interrupt for I2C?

I noticed that the I2C library doesn't make use of interrupts, and even the register pulling mode doesn't actually make use of control registers, I left this issue open: https://github.com/tenbaht/sduino/issues/143

I was able to make write in I2C following the example code that ST provided, but the part of reading I2C I couldn't implement.

I'd like to try the I2C implementation with interrupt, but I'm finding examples only for the Cosmic compiler, which unfortunately is a barrier for enforcing licensing via email.

How can I make the attachInterrupt function direct I2C interrupts to the handler?

Ref.: https://github.com/tenbaht/sduino/issues/143

kokospalme, 2024-04-16:
When I try to implement [this example-sketch](https://tenbaht.github.io/sduino/usage/interrupts/#parameters):
```
// the pin where the input button is attached. Change, if needed
#define BUTTON  PA2

// volatile is important, because this variable is modified in IRQ routine
volatile uint8_t flag = 0;

void on_button_pressed(void)
{
    flag = 1;
}

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, 1);   // turn off the LED

    pinMode(BUTTON, INPUT_PULLUP);

    attachInterrupt(digitalPinToInterrupt(BUTTON), on_button_pressed, FALLING);
}

void loop()
{
    if (flag) {
        digitalWrite(LED_BUILTIN, 0);
        delay(300);
        digitalWrite(LED_BUILTIN, 1);
        flag = 0;
    }
}
```
I'm getting this error:
```
/Users/*/Documents/Arduino/STM8_DMX/STM8_DMX.ino:21: warning 112: function 'digitalPinToInterrupt' implicit declaration
/Users/*/Documents/Arduino/STM8_DMX/STM8_DMX.ino:21: error 101: too many parameters 
```
When I change ```digitalPinToInterrupt(BUTTON)```to ```digitalPinToPort(BUTTON)```, like in [this comment](https://github.com/tenbaht/sduino/issues/92#issuecomment-737868393) I'm still getting this error:
```
?ASlink-Warning-Undefined Global '_I2C_IRQHandler' referenced by module 'main'
```

Did I miss something here? Do I have to modify ```Winterrupts.c```?

hmeijdam, 2024-04-16:
Modifying arduino.h and Winterrupts.c is what I did. Please note that [the modifiications](https://github.com/tenbaht/sduino/commit/0ce7e4c381028f6472eb29e8459210cee90b937d) have never been committed to a new release. They were done after the last released version. So you will have to download those files from this repository separately and overwrite your current ones.

rtek1000, 2024-04-17:
I would like to add that the code generated in the SDCC compiler is smaller, and for small UCs the Arduino can be a waste of program memory.

It is possible to compile using Eclipse IDE, example:

https://github.com/rtek1000/W1209-firmware-modified/tree/master/W1209/W1209-firmware-Eclipse

https://github.com/rtek1000/W1209-firmware-modified

leifclaesson, 2024-04-17:
I've done many, many STM8 projects since my last commented - all on the STM8S003F3P6, the one with 8 kilobytes flash and 1 kilobyte RAM. I've actually stopped using sduino because there was just no way to fit everything in the small memory space. I now use ST Visual Develop with the Cosmic C compiler. It's certainly a lot trickier, and it's straight C instead of C++, but my projects now fit in memory. I2C Slave and Serial debug output together? No problem, fits in 4 kilobytes now. :)

rtek1000, 2024-04-17:
The ST Develop program takes up more space than the program compiled with SDCC as well.

See more details here:

https://github.com/platformio/platform-ststm8/issues/60

kokospalme, 2024-04-17:
Thanks for the quick replys. I copied the whole files in the ```core``` folder, including ```Arduino.h``` and ```Winterrupts.c```, but I get the same error. What could ```?ASlink-Warning-Undefined Global '_I2C_IRQHandler' referenced by module 'main'``` mean? Does the compiler still not find '_I2C_IRQHandler' anywhere? 
And has anyone played arround with getting data over Serial with [this exmaple](https://github.com/tenbaht/sduino/blob/development/examples/uart-int/uart-int.c)?

rtek1000, 2024-04-17:
Hi, for STM8, I suggest using SDCC instead of the Arduino IDE, or Cosmic, or Platformio.

STM8 is very basic, and this link has lots of tips on how to get to know STM8 and an example of a project with the Eclipse IDE:

https://github.com/rtek1000/W1209-firmware-modified

kokospalme, 2024-04-26:
Thank you for the hints, I got it working on PIO, but I don't know if I'm using the interrupts right... see [this thread](https://github.com/platformio/platform-ststm8/issues/62). Can someone help me with this and see if there is any big mistake I make?

commit 0ce7e4c3 (2020-10-30): fix #92: implement attachInterrupt() and detachInterrupt()