tenbaht/sduino #97: Port PG is not working for stm8disco.

ptnicu, 2020-05-18:
Due to missing the PF port the LUT tables (port_to_####_PGM) has wrong place for PG port. PG = 7, but its index in the table is 6.
After adding PF port, the pins on PG port are working fine:
const uint16_t PROGMEM port_to_mode_PGM[] = {
	NOT_A_PORT,
	GPIOA_BaseAddress+2,
	GPIOB_BaseAddress+2,
	GPIOC_BaseAddress+2,
	GPIOD_BaseAddress+2,
	GPIOE_BaseAddress+2,
        **GPIOF_BaseAddress+2,**
	GPIOG_BaseAddress+2,
};

EG-bit, 2020-09-05:
This solution didn't work for me.
Can you write in more detail where you changed these lines of code?

ptnicu, 2020-09-28:
Sorry for not being more specific. 
In the file sduino/stm8/variants/stm8sdisco/pins_arduino.h lines 228-235 the GPIOF is missing for stm8 discovery board.

Also the GPIOF port is missing from port_to_output_PGM, port_to_input_PGM declarations.
Hope this helps.